prusa3d/PrusaSlicer #10573: Solid infill has different patterns at the same time

Matszwe02, 2023-05-15:
### Description of the bug

Printer does normal rectalinear (monotonic) infill, but it combines it with concentric infill at some places (probably narrow ones?)
[image]

### Project file & How to reproduce

[ER.zip](https://github.com/prusa3d/PrusaSlicer/files/11481586/ER.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-beta1

### Operating system

Windows 10

### Printer model

Voron 0

m-sundman, 2023-05-17:
This is unfortunately still present in 2.6.0-beta2, and it's driving me crazy!

riccalisci, 2023-05-23:
same problem

[image]
[part.zip](https://github.com/prusa3d/PrusaSlicer/files/11543930/part.zip)

m-sundman, 2023-05-30:
Still present in 2.6.0-beta3

Matszwe02, 2023-05-31:
It may be useful however, because it appears on parts where normal infill would do short moves. But it needs to be toggleable.

Godrak, 2023-05-31:
Yes, this is intentional. We use Rectilinear, except for regions where a bunch of subsequent lines would be very short, there we use concentric. This reduces vibrations during internal solid printing, which is usually faster. And especially avoids the zigzag pattern on overhangs and near perimeters, which caused melting, curling and patterns on the surface.

Did you encounter any issue, apart from the fact that it may look slightly odd in the preview?

foreachthing, 2023-05-31:
> Yes, this is intentional. We use Rectilinear, except for regions where a bunch of subsequent lines would be very short, there we use concentric.

Was that documented? I must have missed it, too. But this is actually a great "feature"! Are the settings (shortness of lines) hardcoded?

Matszwe02, 2023-05-31:
yeah, if that would be helpful, I'd like to tweak the distance (as well as the toggle I said before)

m-sundman, 2023-06-03:
Non-monotonic fill patterns may be swapped to prevent short distances or what not, but _monotonic_ is monotonic for a reason, and swapping out part(s) of a monotonic fill pattern makes **the layer** **non-monotonic**. Doing non-monotonic infills when monotonic is requested is simply unacceptable, and the sheer arrogance of giving the user the finger while silently doing the opposite of what is asked for is just unbelievable. At the very, very least one must be able to turn off such shenanigans.

Tupson444, 2023-06-04:
I agree, it's in general a bad thing to assume that something is better for everyone and in every application (because most of the time it's an incorrect assumption). We need a toggle, or otherwise the new "feature" may be more harm than useful.

neophyl, 2023-06-04:
Question, to the people talking about monotonic, on your model(s) is the very top layer (the red topfill) not being printed in monotonic then ?  Because from what I can see the new algo is being applied to the normal rectilinear fill but the top layer is still monotonic.  
Monotonic has always just applied to the very last layer (topfill) and not the layers below (there are even other feature requests to change that behaviour).  So practically speaking what difference does it make how its doing the layers below ?  It wasn't mono before.
Of course I don't have the models you are slicing so I'm curious if its applying this new method to topfill too for you.

Tupson444, 2023-06-04:
My bad for now checking that detail, I checked now - it is not applied to top layer, so it's not visible.

Don't get me wrong, I do like the new feature, but it would be better if there was an option to choose in case some downsides show up.

Is there some reason why it's difficult to have a toggle (e.g. if the old algorithm is incompatible with some other part of the code or something)?

m-sundman, 2023-06-06:
I usually use height range modifiers to change how parts are sliced, e.g. with top layers set to 0 and bottom layers set to 999, or vice versa, to make sure that a partial top layer gets filled completely, or other similar tricks. This "smart" feature is messing with such workarounds for the slicer's other deficiencies.

I'm not saying the feature is bad, because it is in fact **great**, but **only if I can turn it off/on when (un)wanted!**

Strilock, 2023-12-27:
This is also very bad because it is causing issues like crazy amounts of unnecessary stringing for PETG. Instead of having 2-4 disconnect points
[image]

 you get about 10 in a small flat part.
[image]

It is also modifying the width of the extrusions.
[image]

Which again is an extra pain point that causes unnecessary stringing that the user did not concent to. This gets caught by a nozzle and you get degraded burnt filament embed in your white print. I don't know what imaginary battles this was meant to combat "pillow effect" or what not, but this is causing real issues. You get a covered nozzle by the end of the print and automatically expect warping, but there is no warping just stringing. This has to be controlled via the infill settings, either make a separate one for the 100% infill like SuperSlicer does or make it behave as prescribed in the top fill.

jmerc77, 2024-01-07:
just noticed this in 2.7.1. the part was quite small and only did concentric on inside of the part.

[image]

paolopix, 2024-02-17:
when making flat prints and in which aesthetics is also important in the layers below the upper one, a filling like this is visible and very annoying. The ideal would be to be able to choose whether to activate or deactivate this function.
[image]
[image]
[image]
[image]

excieve, 2024-11-08:
I'm having a lot of troubles with this when using PETG. It was driving me insane trying all kinds of options until I found this issue.

Everything is great on the bottom and top layers where this isn't happening, but on some models anything in between that has a solid infill is a mess. Less worried about stringing due to this, and more about filament building up on the nozzle and later going places it shouldn't or smearing to neighboring lines potentially causing a broken print.

More Z offset, enabling lift height and reducing speed on solid infill helps somewhat, but on models where it ends up changing the pattern in many parts or on small features, issues are just unavoidable.

Please add an option to disable this or control the minimum area/line size where a pattern change could happen. It would greatly improve the print quality with PETG.

excieve, 2024-11-10:
In OrcaSlicer switching "Detect narrow internal solid infill" option off does exactly this — uses the configured solid infill pattern everywhere, never switching to the concentric pattern in narrow spaces. This makes PETG behave perfectly on solid infill and improves the overall quality of the print as the result.
[image]

itszeekay, 2025-03-26:
Voting on this as well, secondary layers have cause unevenness and even triggered stuck filament detection as it tries to do parallel lines in a long thin wall, the middle line being slightly over-extruded. This could potentially be adjusted with flow rates, but it affects other visuals of my parts as well. We should be able to toggle this option if nothing else. The short moves are less efficient, but needed occasionally. Perhaps configurable by area or minimum size.

MrDixioner, 2025-10-08:
Why can't this feature be disabled? If solid fill was uniform in earlier versions and you removed it for some reason in newer versions (without asking users), is it really that hard to turn it back on as a toggle? Those who need it can use the new feature, but I find it really annoying and it significantly reduces print quality.

Tupson444, 2026-09-18:
a toggle is still wanted

MrDixioner, 2026-09-25:
This remains relevant. Instead of deferring the fix for this function to version 3.0, you should update the previous version (2.9.6), because it is unknown when version 3 will be released or how well it will be received by users. It also seems odd to reclassify this issue as "obsolete"—perhaps it shouldn't have been implemented in the first place, or a way to disable the function should have been provided.

Jan-Soustruznik, 2026-10-06:
Duplicate of #2147

closed 2026-10-06 by Jan-Soustruznik