prusa3d/PrusaSlicer #10575: 2.6.0-beta1 Crashes

1John419, 2023-05-16:
### Description of the bug

Attached stl crashes beta1 when slicing. Same file (and settings) slice with no problem in 2.6.0-alpha6.
[tape_winder_bottom_5.zip](https://github.com/prusa3d/PrusaSlicer/files/11483679/tape_winder_bottom_5.zip)

### Project file & How to reproduce

Open stl with 0.20 QUALITY, Generic PETG, Original Prusa MINI & MINI+. Slice now. PrusaSlicer crashes with no error message.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-beta1

### Operating system

Windows 10 Pro 64-bit 19045.2364

### Printer model

Prusa Mini+

Santoshsomi, 2023-05-16:
crashes when trying to export as stl too.

Jan-Soustruznik, 2023-05-16:
Hello, @1John419 , 

thank you, for you feedback. After check we find that problem is similar to https://github.com/prusa3d/PrusaSlicer/issues/10563

We will try to fix in next release. 
Issue was closed as duplicate

Thank you, have a nice day 

@Santoshsomi  We cannot reproduce the issue with exporting as you report. If you will have any similar issue, please create a ticket. Thank you