prusa3d/PrusaSlicer #10576: Context menu doesn't show on right click of objects on Mac 2.6 Beta 1

pipcorona, 2023-05-16:
### Description of the bug

When selecting multiple objects of and item and right clicking, the context menu does not appear. Instead it deselects the objects and the add shape context menu appears. I tested on version 2.5.2 on the same computer and it works as expected.

### Project file & How to reproduce

https://github.com/prusa3d/PrusaSlicer/assets/3538403/edb3f908-22f1-47c8-a6f6-4ddf4831d707

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 Beta 1

### Operating system

MacOS 13.3.1

### Printer model

MK3S+ MMU2S

Jan-Soustruznik, 2023-05-17:
Hello, @pipcorona, 

thank you for your feedback, problem can be reproduced. We will try to fix. 

Issue created: SPE-1713

FidelCapo, 2023-05-22:
This issue should be fixed and fix will be part of next release.
Closing.

closed 2023-05-22 by FidelCapo