prusa3d/PrusaSlicer #10578: estimated printing time error

riccalisci, 2023-05-16:
### Description of the bug

when in filament overrides there ise't value on detretraction and retraction value, estimated printing time give a same negative big value. Otherwise everything works

[image]
[image]

### Project file & How to reproduce

-uncheck retraction and/or deretraction in filament overrides
-click on "slice now"

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6 beta 1

### Operating system

windows 11

### Printer model

Flsun Q5

enricoturri1966, 2023-05-16:
@riccalisci
Could you please share the 3mf you are using to be sure to repro with the proper settings ?

riccalisci, 2023-05-16:
I used the bunny in shape gallery, but in everything shape that's I used found same problem.

enricoturri1966, 2023-05-16:
I am not talking about the model, but about the config settings (printer, filament, ...).
I am unable to repro with the settings I usually use.

bubnikv, 2023-05-16:
In case the filament override value is not defined (disabled with the check box), one should not access the variable from the script. 
In PrusaSlicer 2.6.0-beta1 a new function is available: is_nil(variable_name), which returns true in case the checkbox is off. In that case one shall use the value defined at printer profile.

We should however extend PrusaSlicer to give an error in case a nil value is accessed.

Jan-Soustruznik, 2023-05-18:
Issue created SPE-1710

bubnikv, 2023-05-18:
PrusaSlicer 2.6.0-beta3 will display the following error message and stop G-code export with your start & end custom G-codes:

[image]

You should use retract_speed[0] instead of filament_retract_speed[0] in case is_nil(filament_retract_speed[0]) is true.

You may find the following useful for working with such complex expressions as above:
https://github.com/prusa3d/PrusaSlicer/wiki/PrusaSlicer-Macro-Language#custom-variables-4048-7196

closed 2023-05-18 by bubnikv