prusa3d/PrusaSlicer #10583: Slicing Model Crashes 2.6 Beta 1

tommck, 2023-05-16:
### Description of the bug

Slicing my model crashes PrusaSlicer

### Project file & How to reproduce

[Bug.zip](https://github.com/prusa3d/PrusaSlicer/files/11489562/Bug.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.0-beta1+win64

### Operating system

Windows 11

### Printer model

MK3S

billyd60, 2023-05-16:
Me too. I've attached another one that crashes 2.6 beta1 when trying to slice it. Slice fine on previous versions.
[vcore3_filament_cutter_orbiter2.zip](https://github.com/prusa3d/PrusaSlicer/files/11489949/vcore3_filament_cutter_orbiter2.zip)

billyd60, 2023-05-16:
> > Me too. I've attached another one that crashes 2.6 beta1 when trying to slice it. Slice fine on previous versions. Win 64, win 10 pro
> > [vcore3_filament_cutter_orbiter2.zip](https://github.com/prusa3d/PrusaSlicer/files/11489949/vcore3_filament_cutter_orbiter2.zip)
> 
> Yeah, mine was a ton of small pieces. Once I combined them and exported them from Fusion 360, it sliced ok.
> 
> Also, to note: 3D Builder detected problems with the model when PS didn't

Ok I ran both stl's through netfabb and it now slices in beta1 without failure. But it didn't say either model had errors, and it slices without using netfabb in prior versions. Interesting!

fersingb, 2023-05-16:
I'm having the same issue with the full size model here: https://www.printables.com/model/27343-burr-puzzle-parametric/files

Version 2.6.0-beta1-GTK3 on ArchLinux (Appimage)

foreachthing, 2023-05-16:
Opened and sliced files with my Ultimaker 2 standard profile:
* "bug" crashed on my system too ( at 45% and infill).
  * HOWEVER, when rotating the model by 90° around X (in Blender or on the other side in PS), it will slice! Rotating back in PS, will result in a crash.
* "vcore3_filament_cutter_orbiter2" sliced with no issue.
* "27343-burr-puzzle-parametric" sliced with no issue.

PS, 2.6b1
Windows 11

fersingb, 2023-05-17:
beta2 fixes it for me

Jan-Soustruznik, 2023-05-18:
Hello, @tommck , 

could you check with beta2 if problem solved, please? 

beta2 release link: https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-beta2

thank you

billyd60, 2023-05-18:
> > > Me too. I've attached another one that crashes 2.6 beta1 when trying to slice it. Slice fine on previous versions. Win 64, win 10 pro
> > > [vcore3_filament_cutter_orbiter2.zip](https://github.com/prusa3d/PrusaSlicer/files/11489949/vcore3_filament_cutter_orbiter2.zip)
> > 
> > 
> > Yeah, mine was a ton of small pieces. Once I combined them and exported them from Fusion 360, it sliced ok.
> > Also, to note: 3D Builder detected problems with the model when PS didn't
> 
> Ok I ran both stl's through netfabb and it now slices in beta1 without failure. But it didn't say either model had errors, and it slices without using netfabb in prior versions. Interesting!
> 
> Edit: Went back and looked and actually one of the models was listed as having 3 shells no errors, but after netfabb became one shell. So the model was to blame, and beta1 seems to be more sensitive to multiple shells in a single model.

Wanted to mention that beta2 now slices this without running the models through netfabb

tommck, 2023-10-04:
> Hello, @tommck ,
> 
> could you check with beta2 if problem solved, please?
> 
> beta2 release link: https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-beta2
> 
> thank you

Sorry it took me an eternity, but this is working fine now.  Thanks

closed 2023-10-04 by tommck

lukasmatena, 2023-10-05:
@tommck Thanks for getting back to us.