prusa3d/PrusaSlicer #10586: Purge tower - inital layer printed inside out - unreliable

viesturz, 2023-05-16:
### Description of the bug

The initial layer of the purge tower is printed from inside out.

For toolchangers with small amount of purge, if the extruder is not perfectly primed, the center of the initial layer can be without filament, causing the purge tower to become unstable or even fail completely.

### Project file & How to reproduce

Print with a purge tower.
Set the purge amount to 15mm3 (the default)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-beta

### Operating system

Win11

### Printer model

voron 2.4

viesturz, 2023-05-16:
Forgot to mention the obvious solution - print it outside-in.

viesturz, 2023-05-16:
[image]

rtyr, 2023-05-16:
You should be able to solve this issue with the new feature/parameter. The size and shape depends on the angle.

[image]

[image]

viesturz, 2023-05-17:
Thanks, that definitely helps with my immediate problem.
Still reversing the order would be an improvement.

jettoblack, 2023-09-04:
I'm having the same issue and the stabilization cone is a workaround but doesn't totally solve the issue, a much better solution would be to reverse the order of printing the wipe tower brim and inside part, and/or print the wipe tower AFTER the (main print) skirt instead of before, so the skirt helps prime the nozzle.

viesturz, 2024-06-16:
Any updates on this one? 

The stabilization cone just about works for taller prints, but it completely useless on small, up to 20mm tall prints. The initial layer just has empty space under the wipe tower.

The prime all extruders: wipe tower is generally the last thing to be printed during first layer, using the last filament in that layer, which give it plenty of time to ooze out underxtrude in the crucial tower base location.

Also prime all exctruders forces a fixed priming pattern and does not take into account how much each nozzle was retracted during previous print

closed 2026-09-29 by prusaslicer-bot[bot]