prusa3d/PrusaSlicer #10587: Slicer gives false data error when extrusion width is set in percentige

Mikelodoncz, 2023-05-16:
### Description of the bug

I´ve specified extrusion width in percents and it gives error, when i write it back to milimeters the error is gone. 

### Project file & How to reproduce

[ps error screenshot.zip](https://github.com/prusa3d/PrusaSlicer/files/11491853/ps.error.screenshot.zip)
[VoronDesign Mobius-Extruder m4 STLs.zip](https://github.com/prusa3d/PrusaSlicer/files/11491858/VoronDesign.Mobius-Extruder.m4.STLs.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 Beta1

### Operating system

windows 10 64bit

### Printer model

TwoTrees Sapphire Pro

neophyl, 2023-05-17:
You do realise that extrusion widths when set to percentages are based on your Layer Height and NOT your nozzle diameter ?  As such its is correctly warning you about the too thin extrusion produced.

jdlawley, 2023-05-20:
neophyl, thank you for clarifying that. Turns out the bug I came to report is actually a documented feature! I don't understand the purpose of setting extrusion width as a function of layer height, when the driving variable should be nozzle diameter.

Also, it doesn't always yield the error. I noticed that I get an error when I set a width to 100%, but not above that. I am running a 1.0mm nozzle with a layer height of 0.69mm. Anything under 145% should result in a too-thin extrusion. My first layer is 0.2mm, set at 105% extrusion width - I got these tiny lines instead of an error that in theory should popup at values under 500% (or thereabouts) for that combination.

[image]

[Caulk Gun Adapter Bushing - extrusion width comparisons.zip](https://github.com/prusa3d/PrusaSlicer/files/11522839/Caulk.Gun.Adapter.Bushing.-.extrusion.width.comparisons.zip)

closed 2023-05-22 by Mikelodoncz