prusa3d/PrusaSlicer #10588: core dumped on 2.5 when complete individual objects is enable

hl037, 2023-05-16:
### Description of the bug

When complete individual objects is enable I get a coredump preceded with a STL assertion error

### Project file & How to reproduce

[pieces.zip](https://github.com/prusa3d/PrusaSlicer/files/11492399/pieces.zip)

Open the project, and press "slice",

I get this error:

```
/usr/include/c++/12.2.1/bits/stl_vector.h:1123: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = double; _Alloc = std::allocator<double>; reference = double&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
[1]    22716 IOT instruction (core dumped)  prusa-slicer
```

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.2 linux

### Operating system

archlinux latest

### Printer model

ender 3 max neo

J3r0github, 2023-05-19:
Unable to reproduce

hl037, 2023-05-22:
Core dumps may very well occur only on specific os since the stdlib implementation differs

closed 2026-09-29 by prusaslicer-bot[bot]