prusa3d/PrusaSlicer #10589: 2.6.0 Beta 1 - Sequential Printing Issue

DaleMckeown, 2023-05-16:
### Description of the bug

In 2.6.0 beta1, sequential printing no longer seems to be functioning properly. In 2.6.0 alpha6, I did not have any issues, and could print the same item 8 times in one sequential run.

I can no longer print two of the same items on a 300x300x300 print capacity.

It keeps giving a "Some items are too tall and cannot be printed without extruder collisions." error, despite the objects being on opposite ends of the bed.

[image]

The print is only 27.5mm tall! There is no chance of a collision.

### Project file & How to reproduce

[Config.zip](https://github.com/prusa3d/PrusaSlicer/files/11492581/Config.zip)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 beta1

### Operating system

Windows 10

### Printer model

Creality S1 Plus

rtyr, 2023-05-16:
Expected behavior. Both of your objects are 55mm tall, while you have height clearance set to only 25mm. You will be able to slice it if you increase the height clearance value. The collision check is currently very basic, it does not matter whether objects are placed next to each other (not printable) or like this (printable). It's not ideal, it is currently mainly up to the user to set up the positions and print order properly. 

We already have duplicate issues related to this topic (labeled "sequential print"), so I am closing this as a duplicate.

closed 2023-05-16 by rtyr

DaleMckeown, 2023-05-16:
Thanks for the info. What's weird is I could print 8 of these sequentially
using the same settings and same height clearance in the last alpha, so
this behaviour in beta1 is confusing. I'll play around with the height
clearance tomorrow.

rtyr, 2023-05-16:
There was a bug in previous 2.6.0-alpha, height clearance setting was ignored.