prusa3d/PrusaSlicer #10597: Color change - prints at the wrong height after changing the filament

GodricCZ, 2023-05-17:
### Description of the bug

in the part where the text (symbol) is inserted via the add text function after the filament change is set, the printer exchange will not return to the original height. G-code generated in version 2.6 beta 1. 

g-code from the same file generated in 2.6 alpha 6 is fine

all printed on Prusa MK3S+, layer 0,15mm

### Project file & How to reproduce

[image]

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-beta1+MacOs-arm64

### Operating system

macOS Ventura

### Printer model

Prusa MK3S+

rtyr, 2023-05-17:
> the printer exchange will not return to the original height.

This is handled by the firmware, not by slicer. But please provide your 3MF project file and both g-codes.

GodricCZ, 2023-05-18:
I have FW version 3.12.2-5713

The files are attached:
[3mf and g-code.zip](https://github.com/prusa3d/PrusaSlicer/files/11504878/3mf.and.g-code.zip)

rtyr, 2023-05-18:
Thanks.

PrusaSlicer only emits M600 command, the rest is handled by the firmware.  There is no difference between those two provided gcodes in this regard (the M600 command is at the same place).

I was also not able to reproduce it on MK3S with FW3.12.2 with your g-code.

Please try again and if it happens again, report it at https://github.com/prusa3d/Prusa-Firmware/issues. You can also contact our tech-support via live chat at www.prusa3d.com.

[image]

closed 2023-05-18 by rtyr