prusa3d/PrusaSlicer #10598: In beta 2, I think there might be an incorrect calcualtion for the amount of filament used.

simpat1zq, 2023-05-17:
### Description of the bug

So I wanted to see how much filament a particular layer was using. So I did a color change on that layer, and the next layer, and the layer after that. This way I could see the amount of filament used by 2 of the layers and compare them.(Let's call the lower layer A and the higher layer B) 

For the purposes of keeping it simple, this is a 3x5 gridfinity storage box, and both layers are basically the same area. Layer A is right after the infill is complete, and starting the solid layers. Layer B is the solid layer. From what I'm seeing by looking at the flow rates for both layers, Layer A should use a good bit less infill. However, the gcode preview shows that Layer A is using 12.72g, but layer B is only using 9.47g. 
[3x5x2tall8bins.zip](https://github.com/prusa3d/PrusaSlicer/files/11499603/3x5x2tall8bins.zip)
[image]
[image]
[image]

### Project file & How to reproduce

I have included the project file. There are also 3 screenshots. The first screenshot shows layer 20, which is the last sparse infill layer. Next is layer 21, which is supposed to be a less dense infill. And then layer 22, which is the solid infill layer. As you can see, layer 21 should be much lighter, but is actually heavier. 

I also included screenshots showing the volumetric flow rate. As you can see, the flow rate of almost all of layer 21 is around 10mm^3. And all of layer 22 is 22.4mm^3. Just from that, it seems that layer 22 should be twice the weight of layer 21. 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 b2

### Operating system

windows 11

### Printer model

voron 2.4 afterburner

simpat1zq, 2023-05-17:
Now sure why, but the flow rate screenshots didn't get in there. Here they are:
Layer21
[image]
Layer 22
[image]

murk-sy, 2023-05-18:
Keep in mind that while layer 21 is ~1/2 the volumetric flow rate of layer 22, it is also printed at ~1/4 of the speed.

At first glance, everything seems in order

neophyl, 2023-05-18:
Its quite simple.  Layer 21 is bridge infill and layer 22 is normal infill.  You have thick bridges enabled.  Thick bridges uses the old algo where the layer height is the same as nozzle diameter, as such the layer is printed at 0.6mm rather than the normal layer height of 0.32.  It doesnt weigh in double as its not a solid layer as you pointed out.

As such no bug here, just the result of your chosen settings.

simpat1zq, 2023-05-18:
I tried it with and without thick bridges. The weight of each layer is still the same.

closed 2026-09-29 by prusaslicer-bot[bot]