prusa3d/PrusaSlicer #10599: Cannot add text, no option in the Right Click list to "add part"

allenj444, 2023-05-17:
### Description of the bug

Cannot add text, there is no longer an option to "Add part"

### Project file & How to reproduce

[box.zip](https://github.com/prusa3d/PrusaSlicer/files/11500077/box.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0 Beta 2 Win 64

### Operating system

Windows 11

### Printer model

Ender 3 S1 Pro

Jan-Soustruznik, 2023-05-17:
Hello, @allenj444 , 

Could you check what View mode do you use? In "Simple mode" is not possible to add the text as a part.  

See the knowledge base, link: https://help.prusa3d.com/article/simple-advanced-expert-modes_1765

allenj444, 2023-05-18:
Ahh, that was the problem, thank you for the quick reply.

Question, is there an option to have more than one line of text?

 

Thank you,

Allen

Jan-Soustruznik, 2023-05-18:
Hello, @allenj444 , 

Sure, you during editing text you can do another line by "Enter" button.  It is the same like any text editor.

have a nice day

closed 2023-06-15 by Jony01