prusa3d/PrusaSlicer #10605: PrusaSlicer crashes during slicing

MrQuallzin, 2023-05-17:
### Description of the bug

In the past couple days, PrusaSlicer has started crashing during slicing, even on simple models. I haven't made any major changes that I'm aware of, and am unsure what would be causing this. It seems to crash when Slicing Progress is at 70% (Generating Support Material) even if there is no support material to generate. 
[image]

### Project file & How to reproduce

[WednesdaySign.zip](https://github.com/prusa3d/PrusaSlicer/files/11502488/WednesdaySign.zip)

To reproduce, open the file and attempt to slice. 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-beta1-+win64

### Operating system

Windows 11

### Printer model

Elegoo Neptune 2S/

JohnOCFII, 2023-05-17:
Have you tried Beta-2 yet?  It was released today.  https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-beta2

MrQuallzin, 2023-05-17:
Just downloaded and retested, the error persists at the same slicing percentage.

JohnOCFII, 2023-05-18:
I tested this with PrusaSlicer 2.6.0-b2 on my MacBook Pro (M1) running macOS 13.3.1(a).  If I import just the model from the 3F, it slices correctly and quickly.  If I import as a project, it fails shortly after I start to slice.  Here is the first bit of the crash dump:

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               PrusaSlicer [98702]
Path:                  /Applications/Other Apps/PrusaSlicer-b2.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.6.0-beta2+MacOS-x64-202305170918 (PrusaSlicer-2.6.0-beta2+MacOS-x64-202305170918)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-05-17 22:41:07.9804 -0500
OS Version:            macOS 13.3.1 (22E772610a)
Report Version:        12
Anonymous UUID:        E1A9B3E8-17C0-3635-633F-31D076EC9805

Sleep/Wake UUID:       66EAD22E-5E9D-4FF5-92CA-D407136AA820

Time Awake Since Boot: 370000 seconds

System Integrity Protection: enabled

Crashed Thread:        15  slic3r_tbb_3

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000048
Exception Codes:       0x0000000000000001, 0x0000000000000048

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [98702]

VM Region Info: 0x48 is not in any region.  Bytes before following region: 105553518919608
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      MALLOC_NANO (reserved)   600018000000-600020000000 [128.0M] rw-/rwx SM=NUL  ...(unallocated)

Thread 0:: slic3r_main Dispatch queue: com.Metal.CommandQueueDispatch
0   libsystem_kernel.dylib        	       0x18bfbff14 mach_msg2_trap + 8
1   libsystem_kernel.dylib        	       0x18bfd2240 mach_msg2_internal + 80
2   IOKit                         	       0x18f1867dc io_connect_method + 516
3   IOKit                         	       0x18f186570 IOConnectCallMethod + 176
4   IOGPU                         	       0x1a8d0cccc IOGPUCommandQueueSubmitCommandBuffers + 184
5   IOGPU                         	       0x1a8cfd10c -[IOGPUMetalCommandQueue _submitCommandBuffers:count:] + 352
6   IOGPU                         	       0x1a8cfcf84 -[IOGPUMetalCommandQueue submitCommandBuffers:count:] + 72
7   Metal                         	       0x195453d34 -[_MTLCommandQueue _submitAvailableCommandBuffers] + 492
8   Metal                         	       0x19545ad04 __40-[_MTLCommandQueue submitCommandBuffer:]_block_invoke + 28
9   libdispatch.dylib             	       0x18be50400 _dispatch_client_callout + 20
10  libdispatch.dylib             	       0x18be5f97c _dispatch_lane_barrier_sync_invoke_and_complete + 56
11  Metal                         	       0x19545acc0 -[_MTLCommandQueue submitCommandBuffer:] + 140
12  AppleMetalOpenGLRenderer      	       0x1250e5bf0 GLDContextRec::flushContext(bool) + 496
13  GLEngine                      	       0x1f0522960 gleUnbindBufferObject + 72
14  GLEngine                      	       0x1f05380a4 gleUnbindDeleteHashNameAndObject + 84
15  GLEngine                      	       0x1f047f340 glDeleteBuffers_Exec + 2264
16  PrusaSlicer                   	       0x1050761e4 Slic3r::GUI::GLModel::reset() + 76
17  PrusaSlicer                   	       0x104f9af48 Slic3r::GUI::GLModel::~GLModel() + 36
18  PrusaSlicer                   	       0x10507ff18 Slic3r::GUI::GLTexture::render_sub_texture(unsigned int, float, float, float, float, Slic3r::GUI::GLTexture::Quad_UVs const&) + 932
19  PrusaSlicer                   	       0x1050855d0 Slic3r::GUI::GLToolbar::render_background(float, float, float, float, float, float) const + 388
20  PrusaSlicer                   	       0x105084078 Slic3r::GUI::GLToolbar::render_horizontal(Slic3r::GUI::GLCanvas3D const&) + 292
21  PrusaSlicer                   	       0x104f98f7c Slic3r::GUI::GLCanvas3D::_render_view_toolbar() const + 204
22  PrusaSlicer                   	       0x104f80938 Slic3r::GUI::GLCanvas3D::_render_overlays() + 168
23  PrusaSlicer                   	       0x104f7d634 Slic3r::GUI::GLCanvas3D::render() + 1196
24  PrusaSlicer                   	       0x104f8c878 Slic3r::GUI::GLCanvas3D::_refresh_if_shown_on_screen() + 140
25  PrusaSlicer                   	       0x104f88914 Slic3r::GUI::GLCanvas3D::on_idle(wxIdleEvent&) + 324
26  PrusaSlicer                   	       0x106d5b7f4 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 252
27  PrusaSlicer                   	       0x106d5b4c0 wxEvtHandler::ProcessEventLocally(wxEvent&) + 88
[... 6 lines cut ...]
34  PrusaSlicer                   	       0x106bb4284 wxWindowBase::SendIdleEvents(wxIdleEvent&) + 132
35  PrusaSlicer                   	       0x106af5eec wxAppBase::ProcessIdle() + 148
36  PrusaSlicer                   	       0x106c4f5c0 wxApp::ProcessIdle() + 36
37  PrusaSlicer                   	       0x106d641c4 wxCFEventLoop::OSXCommonModeObserverCallBack(__CFRunLoopObserver*, int, void*) + 124
38  CoreFoundation                	       0x18c0ddac4 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 36
39  CoreFoundation                	       0x18c0dd9b0 __CFRunLoopDoObservers + 532
40  CoreFoundation                	       0x18c0dd0e4 __CFRunLoopRun + 1028
[... cut ...]
51  PrusaSlicer                   	       0x106cbf4d0 wxAppConsoleBase::MainLoop() + 120
52  PrusaSlicer                   	       0x106c4f618 wxApp::OnRun() + 36
53  PrusaSlicer                   	       0x106d128b8 wxEntry(int&, wchar_t**) + 88
54  PrusaSlicer                   	       0x1050c2ec0 Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 292
55  PrusaSlicer                   	       0x1046be848 Slic3r::CLI::run(int, char**) + 18152
56  PrusaSlicer                   	       0x1046c3220 main + 204
57  dyld                          	       0x18bca7f28 start + 2236
```

Jan-Soustruznik, 2023-05-18:
Hello, all, 

thank you for you feedback, we will try to fix in next release
issue created: SPE-1714

bubnikv, 2023-05-26:
Fixed with b0a109304704400f6b6c0d7cdbc1ba83074ba71c

closed 2023-05-26 by bubnikv