prusa3d/PrusaSlicer #10606: 2.6.0 beta 2 not seeing new printer profiles.

nifty1a, 2023-05-17:
### Description of the bug

I don't think PrusaSlicer 2.6 Beta1 crashes when opening Configuration Wizard #10556 has been fixed correctly

Beta 2 doesn't include a new printer in the configuration wizard list, when it's "name.idx", "name.ini" and "name" folder containing image resources are dropped into "C:\Users\xxxxx\AppData\Roaming\PrusaSlicer-beta\vendor" folder as it should do (2.6.0.0 Alpha versions work correctly, as image in zip file shows)

### Project file & How to reproduce

[Screenshot 2023-05-17 232915.zip](https://github.com/prusa3d/PrusaSlicer/files/11502877/Screenshot.2023-05-17.232915.zip)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-beta2+win64

### Operating system

Edition	Windows 11 Home Version	22H2 Installed on	‎13/‎10/‎2022 OS build	22621.1702 Experience	Windows Feature Experience Pack 1000.22641.1000.0

### Printer model

Ankermake M5, Ender3, Ender 5, Lulzbot Taz 4, Renkforce RF100

just-trey, 2023-05-18:
I see this issue as well with some open source profiles. If the .idx file is copied to the cache it resolves the issue but is this intended to change like this?

nifty1a, 2023-05-18:
@just-trey Indeed, it is your AnkermakeCE profile I am trying to add. Worked fine on alpha6, by dropping in vendor folder in roaming/PrusaSlicer-alpha, but won't in the beta1 or beta2.
If I put them in the cache folder direct, where do the bedplate stl, and thumbnail SVG go?

just-trey, 2023-05-19:
@Jan-Soustruznik the profiles in question are here. https://github.com/Ankermgmt/prusaslicer-ankermake-ce-profiles the reason I am not officially submitting is the vendor wishes to take over the management of these profiles officially. I want to share an alternative version that I beleive performs better and has more options. Would there be a recommend approach to solve for this?

just-trey, 2023-05-24:
@nifty1a can you respond if you agree withy comments above?

nifty1a, 2023-05-25:
> @nifty1a can you respond if you agree withy comments above?

I do @just-trey .
From my perspective, this is an unexpected, undocumented change in behaviour between the alpha and the beta, which breaks the long standing method of adding third party printer profiles. I consider that the cache folder tree is for pre parsed files, and don't think people should be having to drop files directly in there.

just-trey, 2023-05-28:
It looks like @lukasmatena's code change resolved the issue. I have only been able to test on Windows.

kocikdav, 2023-05-29:
@just-trey Yes, it should work as it did until alphas. @nifty1a was right, the changes were breaking a long standing method and would only bring troubles.

Please reopen the issue, if there are any troubles with current implementation.

closed 2023-05-29 by kocikdav