prusa3d/PrusaSlicer #10607: No fill in small areas.

Revasar, 2023-05-17:
### Description of the bug

[image]
[image]
Slicer is not filling some areas that should be filled. 

### Project file & How to reproduce

[Test_Cube.zip](https://github.com/prusa3d/PrusaSlicer/files/11502962/Test_Cube.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-beta1+win64

### Operating system

Windows 11

### Printer model

Ender 3 Pro w/klipper and mods

Revasar, 2023-05-17:
Please note, one screen grab is from 2.5.2 and the other is 2.6.0 beta 1.

J3r0github, 2023-05-19:
Commenting here just so I remember to have a look at this tomorrow

Revasar, 2023-05-25:
> Commenting here just so I remember to have a look at this tomorrow 

Any luck?

J3r0github, 2023-05-25:
Yeah, just forgot to comment. I haven't been able to reproduce this but that might be a skill-based issue on my end.

Revasar, 2023-06-15:
Still having this issue with RC 1.

Jan-Soustruznik, 2023-08-04:
Hello, @Revasar , 

thank you for your feedback. We will to fix in next release. 

issue: SPE-1820

thank you, have a nice

closed 2023-08-04 by Jan-Soustruznik

bbnaz12, 2023-08-04:
Great thanks!  Do you know when to expect the next release?

Jan-Soustruznik, 2023-08-22:
@bbnaz12 , Hello, soon.