prusa3d/PrusaSlicer #10608: System Info screen has poor colour choices

bryn51, 2023-05-18:
### Description of the bug

Using Prusa Slicer version 2.5.2 I open help > system Info screen.  
I am unable to read the information because it shows in Black Text on a charcoal (dark grey) background.
Its not possible to alter the color combination.

### Project file & How to reproduce

No project is open[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.2

### Operating system

MacOS Ventura 13.3.1

### Printer model

Prusa Mk3S+ + MMU2S

kocikdav, 2023-05-18:
Would you please try our latest build - [2.6.0-beta2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-beta2)?

bryn51, 2023-05-19:
> Would you please try our latest build - [2.6.0-beta2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-beta2)?

Errrm, not sure I want to do that.  I only use production release builds, far too many FB people complain about issues on the pre-release builds.

neophyl, 2023-05-19:
The alphas and beta install into a separate location so wont affect your production profiles.  They can be run in parallel.  You dont have to use it to slice , you just need to check for your specific issue.

bryn51, 2023-09-17:
Time has passed.  Now I am using PS 2.6.1.  The issue remains.
I suggest that at the very least you could provide the means to allow colour scheme to be customised, and to use an internal color scheme instead of being derived from system settings.

closed 2026-09-29 by prusaslicer-bot[bot]