prusa3d/PrusaSlicer #10612: Adding a modifier that changes infill results in a middle section without infill

JHBonarius, 2023-05-18:
### Description of the bug

I had a part which I wanted to reinforce at the top. So I added a (generic) box modifier and changed the infill pattern and density.
The default infill was adaptive cubic at 15% and the modified infill was gyriod at 25%
What I expected was the default infill to immediately change to the modified infill at the edge box.
Instead the default infill suddenly stops at 39.64mm (layer 169), then at 41.14mm (layer 175) solid infil along the perimeter starts and at 42.14mm (layer 179) the modified infill starts.

This creates a weak point in the print where there's no infill (1.5mm)

### Project file & How to reproduce

[lamp-Fusion001.zip](https://github.com/prusa3d/PrusaSlicer/files/11510479/lamp-Fusion001.zip)
[image]

Just slice it and see the mentioned layers

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-beta2

### Operating system

Windows 10

### Printer model

prusa i3 mk3S+

JHBonarius, 2026-09-22:
What does this mean? 3.0 is still in alpha... ??
_If_ it's fixed in 3.0, you can't close issues until 3.0 is released! Or do you expect everybody to use the alpha (I think you explicitly say it shouldn't be used for production and such.)

I just tried it in 2.9.6 and there this bug is still there.

zandr, 2026-10-02:
The issue is still present in 3.0a12. 

[image]