prusa3d/PrusaSlicer #10620: When Cutting 1 object : all other parts / objects are moved to the build plate

cryoglobe, 2023-05-19:
### Description of the bug

When having multiple objects the project 
When cutting one object it's automatically re-organizing all the objects ( beeing or not on the build area ) in the build plate :( 

[image]

### Project file & How to reproduce

[cut - reorg all.zip](https://github.com/prusa3d/PrusaSlicer/files/11520711/cut.-.reorg.all.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Prusa 2.6 beta 2

### Operating system

Windows 10 

### Printer model

Sovol Sv04

dpraid, 2023-05-22:
Same problem happening Mac os Monterey.  After every cut it arranges all items

YuSanka, 2023-05-23:
Thanks for your report. We are working on it (SPE-1733)

cryoglobe, 2023-05-25:
YuSanka , thanks for the feedback 

Remark : it's only affecting objects / part that are set to "visible"

CommandoBot, 2023-05-26:
Only parts that are flaged "visible" are ending up on the build plate. All invisible parts are freshly alligned alongside as well.

From my understanding no parts / objects should be moved when any other object is sliced.

closed 2023-06-02 by lukasmatena

cryoglobe, 2023-06-05:
Cool it's solved in beta 4  :)  thanks

commit 76fdc207 (2023-06-02): Fix cut gizmo doing a full arrange

fixes #10620