prusa3d/PrusaSlicer #10621: Bambu Lab X1 Carbon

wallyflower, 2023-05-20:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe how it would work**
Be as descriptive as your skill level allows you.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or paint drawing about the feature request here.

wallyflower, 2023-05-20:
Hi
I had the 3S+ with MMU2 and loved it but my buddy really wanted it so I sold it to him really cheap.
I now have the X1 Carbon and it is a very good machine but I will be buying the MK4 Prusa as Prusa printers work better printing 3D lab Print, Eclipson and Plane Print R/C aircraft.I have two questions for you.
One is will you ever be supporting Bambu Lab printers.
Two is your web site says 4 to 6 weeks lead time on MK 4 orders.
Will this wait time be coming down in the near future.
I am not very good tweaking machine settings and that is why I will be buying the MK 4
Thank You for your time
Wally Fowler

murk-sy, 2023-05-20:
Third party printer support is based on community contributions - Prusa themselves do not make profiles for other manufacturers.

As for the other questions, lead time should eventually go down. But for those questions you should contact support, either using chat on the main site or by sending an email to info@prusa3d.com.

closed 2023-05-20 by rtyr