prusa3d/PrusaSlicer #10626: PrusaSlicer crashes after Extruder column is clicked

alphanci, 2023-05-20:
### Description of the bug

I created a multi extruder printer profile with 2 extruders. I add a model and click under Extruder column on the right sidebar for the model to change the extruder assigned to that model. I get Assertion failed pop up window with the following info:

````
ASSERT INFO:
/usr/src/debug/prusa-slicer/deps/destdir/usr/local/include/wx-3.1/wx/strvararg.h(480): assert "(argtype & (wxFormatStringSpecifier<T>::value)) == argtype" failed in wxArgNormalizer(): format specifier doesn't match argument type

BACKTRACE:
[1] gtk_cell_renderer_start_editing
[2] gtk_cell_area_activate_cell
[3] g_signal_emit_valist
[4] g_signal_emit
[5] g_cclosure_marshal_VOID__BOXEDv
[6] g_signal_emit_valist
[7] g_signal_emit
[8] gtk_event_controller_handle_event
[9] g_closure_invoke
[10] g_signal_emit_valist
[11] g_signal_emit
[12] gtk_main_do_event
[13] g_main_context_dispatch
[14] g_main_loop_run
[15] gtk_main
[16] __libc_start_main
````

I click continue, error window appears again, I click continue again and then I can change the extruder. Same window appears if I try to change the extruder again.  After 1 or 2 changes PrusaSlicer crashes with "Segmentation fault(core dumped" output at the terminal. Below is the full terminal output:

````
[alp@archdesktop ~]$ prusa-slicer 
07:11:54 PM: Debug: window wxTreeCtrl(0x562d6f7ee1e0) lost focus even though it didn't have it
07:11:54 PM: Debug: window wxTreeCtrl(0x562d6f7ee1e0) lost focus even though it didn't have it
07:11:56 PM: Debug: window wxTreeCtrl(0x562d6f7ee1e0) lost focus even though it didn't have it
07:11:56 PM: Debug: window wxTreeCtrl(0x562d6f7ee1e0) lost focus even though it didn't have it

(prusa-slicer:2841): Gtk-WARNING **: 19:12:16.676: Failed to fetch network locations: Timeout was reached
/usr/src/debug/prusa-slicer/deps/destdir/usr/local/include/wx-3.1/wx/strvararg.h(480): assert "(argtype & (wxFormatStringSpecifier<T>::value)) == argtype" failed in wxArgNormalizer(): format specifier doesn't match argument type
/usr/src/debug/prusa-slicer/deps/destdir/usr/local/include/wx-3.1/wx/strvararg.h(480): assert "(argtype & (wxFormatStringSpecifier<T>::value)) == argtype" failed in wxArgNormalizer(): format specifier doesn't match argument type

(prusa-slicer:2841): GLib-GObject-CRITICAL **: 19:14:11.411: g_object_ref: assertion 'G_IS_OBJECT (object)' failed

(prusa-slicer:2841): Gdk-WARNING **: 19:14:11.423: Event with type 1890150256 not holding a GdkSeat. It is most likely synthesized outside Gdk/GTK+

(prusa-slicer:2841): Gdk-CRITICAL **: 19:14:11.423: gdk_window_get_screen: assertion 'GDK_IS_WINDOW (window)' failed

(prusa-slicer:2841): Gtk-WARNING **: 19:14:11.423: no display for event, using default

(prusa-slicer:2841): Gdk-CRITICAL **: 19:14:11.434: gdk_window_get_display: assertion 'GDK_IS_WINDOW (window)' failed

(prusa-slicer:2841): GLib-GObject-CRITICAL **: 19:14:11.434: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
/usr/src/debug/prusa-slicer/deps/destdir/usr/local/include/wx-3.1/wx/strvararg.h(480): assert "(argtype & (wxFormatStringSpecifier<T>::value)) == argtype" failed in wxArgNormalizer(): format specifier doesn't match argument type
/usr/src/debug/prusa-slicer/deps/destdir/usr/local/include/wx-3.1/wx/strvararg.h(480): assert "(argtype & (wxFormatStringSpecifier<T>::value)) == argtype" failed in wxArgNormalizer(): format specifier doesn't match argument type

(prusa-slicer:2841): GLib-GObject-CRITICAL **: 19:14:17.710: g_object_ref: assertion 'G_IS_OBJECT (object)' failed

(prusa-slicer:2841): Gdk-WARNING **: 19:14:17.711: Event with type 1881736656 not holding a GdkSeat. It is most likely synthesized outside Gdk/GTK+

(prusa-slicer:2841): Gdk-CRITICAL **: 19:14:17.711: gdk_window_get_screen: assertion 'GDK_IS_WINDOW (window)' failed

(prusa-slicer:2841): Gtk-WARNING **: 19:14:17.711: no display for event, using default

(prusa-slicer:2841): Gdk-CRITICAL **: 19:14:17.714: gdk_window_get_display: assertion 'GDK_IS_WINDOW (window)' failed

(prusa-slicer:2841): GLib-GObject-CRITICAL **: 19:14:17.714: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
/usr/src/debug/prusa-slicer/deps/destdir/usr/local/include/wx-3.1/wx/strvararg.h(480): assert "(argtype & (wxFormatStringSpecifier<T>::value)) == argtype" failed in wxArgNormalizer(): format specifier doesn't match argument type
/usr/src/debug/prusa-slicer/deps/destdir/usr/local/include/wx-3.1/wx/strvararg.h(480): assert "(argtype & (wxFormatStringSpecifier<T>::value)) == argtype" failed in wxArgNormalizer(): format specifier doesn't match argument type

(prusa-slicer:2841): Gdk-WARNING **: 19:14:20.795: Event with type 0 not holding a GdkSeat. It is most likely synthesized outside Gdk/GTK+

(prusa-slicer:2841): Gtk-WARNING **: 19:14:20.795: no display for event, using default
/usr/src/debug/prusa-slicer/deps/destdir/usr/local/include/wx-3.1/wx/strvararg.h(480): assert "(argtype & (wxFormatStringSpecifier<T>::value)) == argtype" failed in wxArgNormalizer(): format specifier doesn't match argument type
/usr/src/debug/prusa-slicer/deps/destdir/usr/local/include/wx-3.1/wx/strvararg.h(480): assert "(argtype & (wxFormatStringSpecifier<T>::value)) == argtype" failed in wxArgNormalizer(): format specifier doesn't match argument type

(prusa-slicer:2841): GLib-GObject-CRITICAL **: 19:14:24.260: g_object_ref: assertion 'G_IS_OBJECT (object)' failed

(prusa-slicer:2841): Gdk-WARNING **: 19:14:24.261: Event with type 1886341472 not holding a GdkSeat. It is most likely synthesized outside Gdk/GTK+

(prusa-slicer:2841): Gdk-CRITICAL **: 19:14:24.261: gdk_window_get_screen: assertion 'GDK_IS_WINDOW (window)' failed

(prusa-slicer:2841): Gtk-WARNING **: 19:14:24.261: no display for event, using default

(prusa-slicer:2841): Gdk-CRITICAL **: 19:14:24.264: gdk_window_get_display: assertion 'GDK_IS_WINDOW (window)' failed

(prusa-slicer:2841): GLib-GObject-CRITICAL **: 19:14:24.264: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
/usr/src/debug/prusa-slicer/deps/destdir/usr/local/include/wx-3.1/wx/strvararg.h(480): assert "(argtype & (wxFormatStringSpecifier<T>::value)) == argtype" failed in wxArgNormalizer(): format specifier doesn't match argument type
/usr/src/debug/prusa-slicer/deps/destdir/usr/local/include/wx-3.1/wx/strvararg.h(480): assert "(argtype & (wxFormatStringSpecifier<T>::value)) == argtype" failed in wxArgNormalizer(): format specifier doesn't match argument type

(prusa-slicer:2841): Gdk-WARNING **: 19:14:27.832: Event with type 0 not holding a GdkSeat. It is most likely synthesized outside Gdk/GTK+

(prusa-slicer:2841): Gtk-WARNING **: 19:14:27.832: no display for event, using default
/usr/src/debug/prusa-slicer/deps/destdir/usr/local/include/wx-3.1/wx/strvararg.h(480): assert "(argtype & (wxFormatStringSpecifier<T>::value)) == argtype" failed in wxArgNormalizer(): format specifier doesn't match argument type
/usr/src/debug/prusa-slicer/deps/destdir/usr/local/include/wx-3.1/wx/strvararg.h(480): assert "(argtype & (wxFormatStringSpecifier<T>::value)) == argtype" failed in wxArgNormalizer(): format specifier doesn't match argument type

(prusa-slicer:2841): GLib-GObject-CRITICAL **: 19:14:31.186: g_object_ref: assertion 'G_IS_OBJECT (object)' failed

(prusa-slicer:2841): Gdk-WARNING **: 19:14:31.206: Event with type 1890034800 not holding a GdkSeat. It is most likely synthesized outside Gdk/GTK+

(prusa-slicer:2841): Gdk-CRITICAL **: 19:14:31.206: gdk_window_get_screen: assertion 'GDK_IS_WINDOW (window)' failed

(prusa-slicer:2841): Gtk-WARNING **: 19:14:31.206: no display for event, using default

(prusa-slicer:2841): Gdk-CRITICAL **: 19:14:31.209: gdk_window_get_display: assertion 'GDK_IS_WINDOW (window)' failed

(prusa-slicer:2841): GLib-GObject-CRITICAL **: 19:14:31.209: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
/usr/src/debug/prusa-slicer/deps/destdir/usr/local/include/wx-3.1/wx/strvararg.h(480): assert "(argtype & (wxFormatStringSpecifier<T>::value)) == argtype" failed in wxArgNormalizer(): format specifier doesn't match argument type
/usr/src/debug/prusa-slicer/deps/destdir/usr/local/include/wx-3.1/wx/strvararg.h(480): assert "(argtype & (wxFormatStringSpecifier<T>::value)) == argtype" failed in wxArgNormalizer(): format specifier doesn't match argument type
Segmentation fault (core dumped)
````

### Project file & How to reproduce

[3DBenchy.zip](https://github.com/prusa3d/PrusaSlicer/files/11522749/3DBenchy.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.5.2 ( I am using the prusa-slicer package in the community repository)

### Operating system

Arch Linux 6.3.2

### Printer model

Jubilee

kubispe1, 2023-05-25:
Hi, thanks for the issue. Could you please help us by testing it in version 2.6.0-beta2? Thank you very much for your cooperation.

alphanci, 2023-05-25:
I downloaded 2.5.2 AppImage(PrusaSlicer-2.5.2+linux-x64-GTK3-202303231201.AppImage) and 2.6.0-beta2 AppImage (PrusaSlicer-2.6.0-beta2+linux-x64-GTK3-202305170854.AppImage). Tested both and this error never happened. Same 2.5.2 version on Arch repo produced this error and this leads me to believe that this issue is caused by wxWidgets library installed on Arch. I am using the AppImage binary now.

closed 2023-05-26 by alphanci