prusa3d/PrusaSlicer #10627: Travel speed and Settings are inconsistent

nmju99, 2023-05-21:
### Description of the bug

prusa2.6β2 travel speed set to 300, but the real Max is only 160. And that speed is extremely unstable. prusa2.5 doesn't have this problem.

### Project file & How to reproduce

prusa2.6β2 travel speed set to 300, but the real Max is only 160. And that speed is extremely unstable. prusa2.5 doesn't have this problem.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-beta2+win64-202305170854

### Operating system

windows10

### Printer model

toolchanger

rtyr, 2023-05-21:
Please provide your 3mf project file

nmju99, 2023-05-21:
[gai.zip](https://github.com/prusa3d/PrusaSlicer/files/11524036/gai.zip)

rtyr, 2023-05-22:
> prusa2.6β2 travel speed set to 300, but the real Max is only 160.

What makes you think that? I don't see such issue with your project file. Travels are using feedrate F18000 (300mm/s) in the gcode. Your FW may limit those speeds though (not related to PrusaSlicer).

[image]

nmju99, 2023-05-22:
Thank you very much. I see that your screenshot does show 300. But Prusprus2.5 would normally move 300 speed to Prusprus2.6 would not. My English is not good, I don't know if you can understand what I said. Same printer, same computer, same setup, prusa2.5 Setup 300. Runs 300. But prusa2.6 runs less than 300.    I would be very happy if you could help me. Now I really hope that the problem of 2.6 changing the position of the print head will be solved first, instead of changing the print head on the model

nmju99, 2023-05-22:
Thank you very much for your reply. I am at work now. I'll upload it when I get back tonight. Thank you again

rtyr, 2023-09-12:
Are you still going to provide some g-codes?

lukasmatena, 2023-11-12:
No response, closing.

closed 2023-11-12 by lukasmatena