prusa3d/PrusaSlicer #10631: Negative Volumes get messed up when cutting an object

CommandoBot, 2023-05-21:
### Description of the bug

I use negative volumes for cutting objects to have an easy way to glue, fix and pose the parts after printing. The connectors you added are a good starting point, but for my large size figures something more rugget that helps fixating is preferred.

Nevertheless, when cutting the figure and have the objects for the cut selected, the negative volumes get messed up. Some parts of it are only in one cut object, the other ones in the other ... instead of beeing in both as intended.

[image]

When I don't select subobjects in between the cut is done as expected:

[image]

### Project file & How to reproduce

1.) Put the separation plane on a height of 370mm (where the negative volumes are positioned
2.) Change all the objects to belong to the upper part except one leg
[image]
3.) Cut the object
4.) Five of the negative volume pins are located in the leg and only one in the remaining body. In total it should be six negative volumes in the leg and the body.
[image]

When not selecting subobjects and just cut in the height of 370mm it works as intended:
[image]
[image]

 
[ShadowBot_Simplf.zip](https://github.com/prusa3d/PrusaSlicer/files/11524940/ShadowBot_Simplf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 alpha 2

### Operating system

Windows 10, 64Bit

### Printer model

N/A

kubispe1, 2023-05-22:
Negative volumes (which means your extra connectors) need to be selected or unselected, but it is not possible...Thanks

CommandoBot, 2023-05-22:
Thanks ... 

The important thing is, that the negative volume needs to stay on both sides ... So it can't be put either in the magenta or cyan section ...

When not selecting separate objects, just cutting with a plane, it works fine for my needs.

kubispe1, 2023-05-22:
Yes, we noticed also that... We are looking into it.

FidelCapo, 2023-05-31:
This issue should be fixed and the fix will be part of the next release.
Closing.

closed 2023-05-31 by FidelCapo