prusa3d/PrusaSlicer #10632: Changing Slice Gap Closing Radius to 0.001 causes a crash

arshish1612, 2023-05-22:
### Description of the bug

If you change the Slicing Gap Closing Radius Prusa Slicer Crashes the moment you click SLICE

[CREASH.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11527299/CREASH.3mf.zip)

### Project file & How to reproduce

JUst use the attached project file where the parameter Changing Slice Gap Closing Radius is changed from default. Clicking Slice should cause the crash

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0 Beta 2

### Operating system

Macbook Air M1 - 13.3.1 (a) 

### Printer model

FLSUN V400

Jan-Soustruznik, 2023-05-23:
Hello, @arshish1612 , thank you for you report. We will try to fix it. 

Issue: SPE-1729 

Have a nice day

lukasmatena, 2024-02-09:
Fixed in [2.7.2-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.2-alpha2). Closing.

closed 2024-02-09 by lukasmatena