prusa3d/PrusaSlicer #10633: Center model on the platter

rociomarielgnas, 2023-05-22:
**Is your feature request related to a problem? Please describe.**
It's REALLY difficult to precisely align different STLs for a print.

**Describe the solution you'd like**
Please add at least a "Center STL" option on the context menu of a model/part. If we can center them together then group them we are golden already.

**Describe how it would work**
Right click a model, select Center STL. It centers it to middle of X and Y of the current build plate size.

**Describe alternatives you've considered**
I set X and Y position on each model as 0 then group them.

**Additional context**
Cura has this option on their context menu. Maybe you guys could check their source for an idea of how they do it.

Mikelodoncz, 2023-05-22:
You can hold down ctrl and than click on the model names displayed on the right and then write coordinates on the bottom right.

rociomarielgnas, 2023-05-22:
> You can hold down ctrl and than click on the model names displayed on the right and then write coordinates on the bottom right.

Yes that's what I do right now. I mentioned it on the issue. But it isn't easy to eyeball coordinates when the things aren't the same width or length. I know I can write half of my plate's size on x and y and that would be centered but it's a pain in the butt. I don't think making a context menu option that does that for you would be that difficult to implement.

ghawk1ns, 2024-10-07:
@Mikelodoncz the coordinates don't seem to be aligned with the build plate - for instance setting an object's world coordinates to `(0,0,0)` removes the object from view along with a `Error: An object outside the print area was detected` notification. It would be nice, (I believe what Cura does this) if coordinates `(0,0,0)` was centered on the build plate. Or if there was another coordinate type, `Build Plate Coordinates`, in addition to `World Coordinates` and `Object Coordinates`