prusa3d/PrusaSlicer #10634: Table heating

atas80, 2023-05-22:
**Describe the solution you'd like**
For serial printing, I want there to be a function to turn off the heating of the table before the end of printing

**Describe how it would work**
Specify the number of layers before the end of printing from which to turn off the table heating

foreachthing, 2023-05-23:
You can easily do that by adding custom G-Codeto the desired height.

[image]
Right-click on that (+) mark.

[image]
Choose "Add custom G-Code".

[image]
Type (or copy/paste) `M140 S0`

Done.

neophyl, 2023-05-23:
Except that you can't add those when using Serial Printing.  The function is disabled.

atas80, 2023-05-25:
Ok. Th.

closed 2026-09-29 by prusaslicer-bot[bot]