prusa3d/PrusaSlicer #10636: [2.6.0 ß2] Filament color selected in Filament menu does not transfer to main screen

koliver1944, 2023-05-22:
### Description of the bug

When I go to the Filament Settings menu and select a new filament color, and then go back to the main screen the color I just selected does not show up. The color remains as orange instead of the color I selected.
I went back and tried it in 2.6.0 Alpha6 and the new color is correctly changed in the main screen.

### Project file & How to reproduce

[Untitled.zip](https://github.com/prusa3d/PrusaSlicer/files/11533061/Untitled.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0 +Beta2 + Win64

### Operating system

Windows 11

### Printer model

Prusa Mini+

YuSanka, 2023-05-24:
Thanks for your report.
We are working on it SPE-1736

FidelCapo, 2023-06-09:
This issue should be fixed and the fix will be part of the next release.
Closing.

closed 2023-06-09 by FidelCapo