prusa3d/PrusaSlicer #10637: PrusaSlicer-2.6.0-beta2+win64 Crashes When Adding Raft Supports

kgrigio, 2023-05-22:
### Description of the bug

When attempting to add raft supports then slicing, the program crashes at 70% or when it is trying to "generate the supports".  I have been able to slice successfully with no raft support, but as soon as I attempt to add raft supports and slice the program crashes completely.

### Project file & How to reproduce

[qx110875.zip](https://github.com/prusa3d/PrusaSlicer/files/11533329/qx110875.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-beta2+win64-202305170854

### Operating system

Windows 10

### Printer model

voron 2.4 with stealthburner

taquitos, 2023-05-22:
Also seeing this crash in macOS 13.4 (22F66)

## Repro steps: 
Open file, press __slice__.
[Playdate Dummy Model.f3d.zip](https://github.com/prusa3d/PrusaSlicer/files/11533864/Playdate.Dummy.Model.f3d.zip)

```
PrusaSlicer
Version:   2.6.0-beta2+MacOS-arm64
Build:     PrusaSlicer-2.6.0-beta2+MacOS-arm64-202305170907

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Ventura Version 13.4 (Build 22F66)
Total RAM size [MB]: 34,360MB
OpenGL installation
GL version:   4.1.0
Profile:      Core
Vendor:       Apple
Renderer:     Apple M1 Max
GLSL version: 4.10.0
Textures compression:       Enabled
<details>
<summary>Installed extensions:</summary>
GL_APPLE_client_storage
GL_APPLE_container_object_shareable
GL_APPLE_flush_render
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_texture_range
GL_ARB_ES2_compatibility
GL_ARB_blend_func_extended
GL_ARB_draw_buffers_blend
GL_ARB_draw_indirect
GL_ARB_explicit_attrib_location
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_occlusion_query2
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_separate_shader_objects
GL_ARB_shader_bit_encoding
GL_ARB_shader_subroutine
GL_ARB_shading_language_include
GL_ARB_tessellation_shader
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_cube_map_array
GL_ARB_texture_gather
GL_ARB_texture_query_lod
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_storage
GL_ARB_texture_swizzle
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_vertex_attrib_64bit
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_texture_compression_s3tc
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_sRGB_decode
GL_NV_texture_barrier
</details>
```

Jan-Soustruznik, 2023-05-23:
Hello, @kgrigio , thank you for your feedback, we will try to fix. 

issue: SPE-1732

bubnikv, 2023-05-25:
Fixed with 1adf916cedc94815b9c37b558faa0a6c7826fa3e

closed 2023-05-25 by bubnikv