prusa3d/PrusaSlicer #10638: Export to SD card, remove card, then Prusaslicer crashed

kocickamia, 2023-05-22:
### Description of the bug

After slicing any STL file, then export G-code to SD card, then click Eject in PS dialog, Windows show system dialog "safely removed hardware". 
[image]
After that, click anywhere or attempt to close Prusaslicer, application crashed and show this:
[image]

This problem was not in 2.6.0-alpha6.

### Project file & How to reproduce

1 - load any stl file
2 - click to Slice
3 - click to Export Gcode (right down corner)
4 - save gcode to SD card
5 - click to Eject SD (or press CTRL-T)
6 - close Prusaslicer
7 - profit...

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-beta2 Win64. 2.6.0-beta1 Win64

### Operating system

Windows 10 Home 64bit 22H2

### Printer model

All of my printers (Kossel linear plus, Mini delta, DIY CoreXY...)

kocickamia, 2023-05-22:
Add info from Eventviewer...
[image]

kocikdav, 2023-05-24:
Hello @kocickamia thank you for reporting this. I'm a bit afraid, this will be a difficult bug to catch. Would you try running the beta  from command line?
- Start Command Prompt
- Navigate to folder with PS 2.6.0 beta (or some systems has option Open in Terminal when right click in file explorer, not sure if Win 10 Home)
- run .\prusa-slicer-console.exe --loglevel=5
- crash it
- copy paste the log here please

Thank you.

kocickamia, 2023-05-24:
Hello @kocikdav , here is the logfile, but last record is "Ejecting has succeed". 
[log.txt](https://github.com/prusa3d/PrusaSlicer/files/11557273/log.txt)

kocickamia, 2023-05-24:
BTW @kocikdav , where are the logfiles from prusa-slicer-console? Not found it. I generated this logfile running "prusa-slicer-console.exe --loglevel=5 > log.txt"

kocickamia, 2023-06-16:
Problem still exist in PS 2.6.0 rc2...

closed 2026-09-29 by prusaslicer-bot[bot]