prusa3d/PrusaSlicer #10639: Variable layer height at certaint places

Mikelodoncz, 2023-05-22:
**Is your feature request related to a problem? Please describe.**
When I want to print part that has a thread or steep overhang and i want to print the thread or overhang in highest quality I have to print the whole part in 0.1mm layer height to get best quality thread or steep overhangs

**Describe the solution you'd like**
It would be awesome if Prusaslicer had feature like selecting threads or steep overhangs and set layer height nad depth from outer perimeter for that thread or overhang for example to 0.1mm layer heght and 3mm depth, but rest of the part could be printed at 0.3mm layer height, thus shortening the print time.

**Describe how it would work**
In 3d editor there would be another card (on the left) like support painting tool, that would enable to paint threads and steep overhangs .

sarusani, 2023-05-22:
https://help.prusa3d.com/article/variable-layer-height-function_1750

Mikelodoncz, 2023-05-22:
> https://help.prusa3d.com/article/variable-layer-height-function_1750

You clearly didn't read what I wrote.

foreachthing, 2023-05-23:
I'd like to have a modifier and be able to set Layer Height there.... Paint-on? Sure, why not.
For mechanical parts this would be a life-saver - I have exactly the same "problems" as Mikelodoncz.

neophyl, 2023-05-23:
You really should search for existing issues before creating new ones.  This is basically the same request as https://github.com/prusa3d/PrusaSlicer/issues/8223

Mikelodoncz, 2023-05-23:
> You really should search for existing issues before creating new ones.  This is basically the same request as https://github.com/prusa3d/PrusaSlicer/issues/8223  

No its not, if I am printing sphere I can add variable layer height modificator to bottom and upper portion, but it changes layer height thru the whole print. If i have a cube 100x100x100mm that has 4 threads at the top that are going thru the whole print I would have to print the whole cube at 0.1mm layer height. With my tool you could print only the threads at 0.1mm and rest of the cube at 0.3mm layer height.

neophyl, 2023-05-23:
That request is to have variable layer heights on separate parts of the same object that do not apply to the whole layer, ie only on the detail parts, your request is to have different layer heights on the same object that dont apply to the whole layer and only on the detail parts.  Why isnt it the same ?  The phrasing is different, the intent is identical.

Mikelodoncz, 2023-05-23:
> That request is to have variable layer heights on separate parts of the same object that do not apply to the whole layer, ie only on the detail parts, your request is to have different layer heights [...]

This part is good example of my requested feature, but you are right.
[image]

closed 2026-09-29 by prusaslicer-bot[bot]