prusa3d/PrusaSlicer #10640: Prints outside the print area

tom-321, 2023-05-22:
### Description of the bug

When I add multiple objects
[image]
 and press "A", the objects are arranged along the boundary of the printing plate,
[image]
[image]
 which causes the skirt to be outside the printing area and an error message is displayed 
[image]

### Project file & How to reproduce

[Körper28.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11536907/Korper28.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.2+MacOS-x64

### Operating system

macOS Ventura Version 13.1 (22C65)

### Printer model

ender3

foreachthing, 2023-05-22:
I don't get it.... it's just too big to print. That's it.
You could try to reduce the skirt distance: Print Settings -> Skirt and Brim -> Distance from Brim/Object.
Or even disable Skirt all together.
Or, manually arrange objects to the fit on the plate with brim. The two hex-box-parts will fit in the hooks when rotated.

foreachthing, 2023-05-23:
Maybe something like this: https://github.com/prusa3d/PrusaSlicer/issues/8442#issuecomment-1399450886

tom-321, 2023-05-23:
Could be a solution if the spacing value is equal or greater than the size of the skirt (or brim)

On the other hand the slicer could just stop the toolpath where the bed edge starts (that's what the current cura version does, if I remember correctly), unfortunately thats not the purpose of the skirt function

Jan-Soustruznik, 2023-05-23:
Hello, @tom-321 , 

I see that you using PrusaSlicer 2.5.2 where isnt implemented the Arrange improvements , that was implemented since in 2.6.0-alpha6 , see the quote from changelog below:

> 'Arrange' and 'Fill bed with instances' functions now take into account skirt and brim separation (https://github.com/prusa3d/PrusaSlicer/issues/3072, https://github.com/prusa3d/PrusaSlicer/issues/3477, https://github.com/prusa3d/PrusaSlicer/issues/5286, https://github.com/prusa3d/PrusaSlicer/issues/5310, https://github.com/prusa3d/PrusaSlicer/issues/9301, https://github.com/prusa3d/PrusaSlicer/issues/9309). Many thanks to @jschuh (https://github.com/prusa3d/PrusaSlicer/pull/7653) and @individ-divided (https://github.com/prusa3d/PrusaSlicer/pull/9181), whose pull requests were very useful sources of inspiration, despite not being used verbatim.

I suggest to check your issue in our latest released beta2, where is all new updates and features already implemented: https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-beta2

I close issue as solved,
Thank you, have a nice day

closed 2023-05-23 by Jan-Soustruznik