prusa3d/PrusaSlicer #10643: Organic Infill

Beta1492, 2023-05-23:
**Is your feature request related to a problem? Please describe.**
After the success of organic supports, I'd like to have an option for organic infill.

**Describe the solution you'd like**
I would like to have Organic as a selectable infill type. 

**Describe how it would work**
(Knowing next to nothing about software) I would assume you could feed the "inside" of the model through the organic algorithm just like organic supports process the "outside" of the model now. 

**Describe alternatives you've considered**
I've tried lightning infill to reduce print time/costs, but find it to be too unstable and messy. 

**Additional context**
Add any other context or paint drawing about the feature request here.

Mikelodoncz, 2023-05-24:
There is already similar thing called lighting infill

Beta1492, 2023-05-25:
I am familiar with lightning, but I've found it to be too unreliable. I'm hoping organic infill would be a perfect balace of reliability, speed, and low material use.

Tupson444, 2026-09-12:
still relevant