prusa3d/PrusaSlicer #10655: PrusaSlicer has encountered a fatal error: "Access violation"

bttlsgtmajor, 2023-05-24:
### Description of the bug

Downloaded and attempted to slice a benchie and get an error `PrusaSlicer has encountered a fatal error: "Access violation"`

### Project file & How to reproduce

[3DBenchy.zip](https://github.com/prusa3d/PrusaSlicer/files/11556363/3DBenchy.zip)
All I've done is loaded up the file and attempted to slice it with one of my profiles I have saved and I get the access violation error upon trying to slice it. I can provide the profile if needed, but I haven't had this issue previously.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-beta2+win64

### Operating system

Windows 11

### Printer model

Sovol SV06

bttlsgtmajor, 2023-05-24:
Looks like this actually may indeed be related to the profile settings.... I tried to slice with one of the other ones I have and it sliced fine. I'm going to see if I can find where the profile is saved so I can upload that.

kubispe1, 2023-05-24:
"I checked it quickly and I think it was caused by a raft. I believe it could be a duplicate of #10637. We will investigate your case along with the previous issue.
[image]