prusa3d/PrusaSlicer #10659: Using less organic structures where supporting multiple objects/instances

einichi, 2023-05-25:
**Is your feature request related to a problem? Please describe.**
When slicing multiple objects, or multiple instances of an object that require supports, Prusa seems to always generate a "tree" for each object or instance of an object when using Organic Supports.

**Describe the solution you'd like**
An option to use only one "tree" base with branches that reach out to multiple objects where permitted by other settings such as preferred/maximum angle. This can potentially improve efficiency with regards to filament usage and print time.

**Describe how it would work**
Consider all objects on the bed when generating organic supports, not just per object. When multiple objects are judged to be within an acceptable distance and the set angle is met, generate branches to other neighbouring objects from a single organic "tree".

**Describe alternatives you've considered**
Beyond avoiding models that require supports in the first place, I'm not sure what alternatives there are.

**Additional context**
A poorly annotated screenshot of what I mean below:

[image]

sarusani, 2023-05-25:
This is more like a workaround than a solution, but it will achieve the desired results:
If you have multiple objects you can use "merge" after you placed them. That will generate the supports like you describe them. 
For instances you first need to use "set as separate objects" and then "merge" again.

einichi, 2023-05-26:
Yep, that workaround works great, thanks @sarusani 

[image]

closed 2026-09-29 by prusaslicer-bot[bot]