prusa3d/PrusaSlicer #10662: Issue with Adding Parts

ZachCondon, 2023-05-25:
### Description of the bug

I'm having an issue with adding parts to print with my MMU2. I have a set of .3mf files that I import my main file first, then add parts to it so that I can print in multiple colors. I checked PrusaSlicer 2.6.0 alpha6 and PrusaSlicer 2.6.0 beta2 and have the same issue. If I go back to PrusaSlicer 2.5.2, the files work as expected.
[image]
[image]

### Project file & How to reproduce

I've included the saved project files from each version as well as the original .3mf files that I used to make each project. I open the lid.3mf file first, then add the rest of the files to the lid in the slicer.
[2.5.2.zip](https://github.com/prusa3d/PrusaSlicer/files/11566443/2.5.2.zip)
[2.6.0.zip](https://github.com/prusa3d/PrusaSlicer/files/11566444/2.6.0.zip)
[v2.zip](https://github.com/prusa3d/PrusaSlicer/files/11566445/v2.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.2+win64 and Version 2.6.0-beta2+win64

### Operating system

Windows 11

### Printer model

Mk3+ with MMU2

Jan-Soustruznik, 2023-05-31:
SPE-1742

lukasmatena, 2023-06-09:
Thanks for the report. Fixed in [2.6.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-rc1). Closing.

closed 2023-06-09 by lukasmatena