prusa3d/PrusaSlicer #10663: checkbox for using (or not) bridge settings on first solid layer over infill

exec79, 2023-05-25:
**Is your feature request related to a problem? Please describe.**
PS uses bridge and fan speed on the first solid layer over the infill. Problem with this, sometimes you need slow bridge speed and high fan for "real" bridges, but you really dont want to print the first solid layer over the infill at 5-10mms with 100% fan especially if the part is printed from a cooling sensitive material like ABS. Currently i have to use modifiers to increase speed, and manual Gcode editing to remove the high fan speeds at layers with unwanted bridging.

**Describe the solution you'd like**
I would like a checkbox or something that can select wether i want or dont want first solid layers over infill considered as bridge.

**Describe how it would work**
print first "over the infill first solid layer" as bridge, or print it with regular solid layer settings

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
[image]

wuast94, 2023-08-27:
Related to this i would also love to see a setting where i can set a threshold distance if its a bridge or not. for example only extrusions midair over 5mm are bridges. this would be a bit more dynamic for different scenarios than just an on off switch

Karl-Huna, 2023-08-27:
Hey there,

same thoughts here.
It would speed up the print preprocessing alot and the print too.
Mostly it is not neccessary to bridge over sparse infill so i am wondering if there could be a checkbox to switch the 
first layer over sparse infill from bridge to solid infill.

closed 2026-09-29 by prusaslicer-bot[bot]