prusa3d/PrusaSlicer #10665: The perimeter is outset of the cross-section before an abrupt transition in cross sectional shape, where top surfaces start.

gnydick, 2023-05-25:
### Description of the bug

The perimeter is outset of the cross-section before an abrupt transition in cross sectional shape, where top surfaces start.

Green arrows in the image show where it looks like the top surfaces start for the section that is becoming a top surface. The protrusion happens all the way around the print. This has happened on multiple printers.

[image]

### Project file & How to reproduce

[solid gantry block v9.zip](https://github.com/prusa3d/PrusaSlicer/files/11569634/solid.gantry.block.v9.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-beta2+win64

### Operating system

Windows 11 Professional 22H2

### Printer model

voron 0.1 and RailCore II ZLT

neophyl, 2023-05-26:
This is a duplicate of many existing open and closed issues here.  
For example please see https://github.com/prusa3d/PrusaSlicer/issues/991#issuecomment-458614931

SpannMagoo, 2023-07-05:
I am seeing this now in Prusaslicer 2.6.0 Release and Orcaslicer.

[image][image][image]

closed 2026-09-29 by prusaslicer-bot[bot]