prusa3d/PrusaSlicer #10677: Solid Infill odd geometry

The-Bishop82, 2023-05-27:
### Description of the bug

Slicing a part at .2 layer height, the solid infill layers display odd geometry. Instead of a rectilinear pattern, it's half rectilinear and half concentric, in an alternating pattern. Visible at layer 184, 183 and so on. Seems to be in all of the smaller solid infill areas.

### Project file & How to reproduce

[GiviSlider-LayerOdd.zip](https://github.com/prusa3d/PrusaSlicer/files/11581944/GiviSlider-LayerOdd.zip)
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-beta3+win64-202305261352

### Operating system

Windows 11 22H2

### Printer model

Highly Modified Ender 3V2, using Ender 3V2 profile

The-Bishop82, 2023-05-27:
Just to add in, Prusaslicer 2.5.2 does not exhibit this behavior.

Godrak, 2023-05-31:
Yes, this is intentional. We use Rectilinear, except for regions where a bunch of subsequent lines would be very short, there we use concentric. This reduces vibrations during internal solid printing, which is usually faster. And especially avoids the zigzag pattern on overhangs, which caused melting, curling and patterns on the surface.

Did you encounter any issue, apart from the fact that it may look slightly odd in the preview? :)

The-Bishop82, 2023-05-31:
No problems with the print, just thought the solid infill layers looked odd as they were different than previous behavior. 

The way you explained it makes complete sense, thank you!

The-Bishop82, 2023-05-31:
Learned that this is expected behavior.

5axes, 2023-05-31:
> Yes, this is intentional. We use Rectilinear, except for regions where a bunch of subsequent lines would be very short, there we use concentric. This reduces vibrations during internal solid [...]

Any possibility to activate/deactivate this new  behaviour via an option ?

secv, 2023-08-05:
In my case, the concentric section overlaps too closely causing a big increase in nozzle pressure, which then spits a huge blobs after the concentric section is complete. There's basically 1 or 2 more perimeters at center than necessary.

Jan-Soustruznik, 2023-08-08:
Hello, @RickM777 , 

Could you share a project file(3mf), please? 

thank you, have a nice day

RickM777, 2023-08-08:
Hi Jan,
I didn't save that project file but I created another one to show the same issue.  I tried to remember which letters cause the most problem but it is a combo of size and how much I have "squished" the letters horizontally and at this size most of the letters look ok.  I'll attach 2 more similar photos and the project file.

[image]
[image]

[Letter Gaps.zip](https://github.com/prusa3d/PrusaSlicer/files/12294478/Letter.Gaps.zip)

As you can see, same issue on layer 2 with the "S" and the "2".  The issue presents in the green parts between letters as well though this is not as much of an issue for us as there will eventually be solid green over the gaps so less color discrepancies.

Interestingly if I switch to a standard 0.2 quality preset print settings the issue is less, though the gap remains in the "2" AND it does show one anomaly at layer 4 in the top "S" which is unusual.

Let me know anything else I can do.
Cheers.

closed 2026-09-29 by prusaslicer-bot[bot]