prusa3d/PrusaSlicer #10679: Build failure on GCC 13.1.1

bluecube, 2023-05-27:
### Description of the bug

When building gentoo package media-gfx/prusaslicer-2.5.2-r1 with GCC 13.1.1, I'm getting build errors:

[build.log.gz](https://github.com/prusa3d/PrusaSlicer/files/11582407/build.log.gz)

I think either most of them (or all?) are caused by missing `#include <cstdint>`.

### Project file & How to reproduce

`sudo emerge prusaslicer`

There's no project file, PrusaSlicer was not built yet.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.2

### Operating system

Gentoo

### Printer model

Ender 3v2

schiele, 2023-05-28:
Does https://github.com/prusa3d/PrusaSlicer/pull/10499 help?

bluecube, 2023-05-28:
It definitely helps I've just tried manual build from source based on the Linux build instructions here in docs, that gets further along than `master`. Unfortunately the build gets stuck later on something else for me and I can't really debug it right now. But I think your PR will fix the build for me once it's merged.

I guess we can close this?

closed 2023-06-22 by bluecube