prusa3d/PrusaSlicer #10682: Organic Support outside of Print Area

Animo78, 2023-05-28:
### Description of the bug

When printing larger objects which expand going up (For example a helmet upside down), the organic support tends to go outside the printing area.

Would expect it to stay in the print area or move the print over a tiny bit, so it would fit.

[image]

### Project file & How to reproduce

Tried it with a new configuration, same issue.
Unfortunately, I cannot share the project because it's a bought (copyrighted) file.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 Beta 2 and 3.

### Operating system

Windows 11 Pro x64

### Printer model

Prusa XL

Cadsters, 2023-05-28:
In 2.6.0 beta 3 you can fix this under: Printer Settings  -> Support material -> Options for support material and raft and then the 4th option from the bottom -> Support on build plate only:
[image]

Animo78, 2023-05-28:
It was a good suggestion.
But It only lifts the object up higher, so support can get under it.
Unfortunately, organic support doesn't go straight up, halfway it is still outside the print area

[image]

Would also be handy if there was an option, where it would move/rotate the print so it would best fit, with the organic support.