prusa3d/PrusaSlicer #10688: [MacOS] "Loading of a model file failed." for .3mf file generated by PrusaSlicer

johanLsp, 2023-05-29:
### Description of the bug

Prusaslicer is unable to load again a project that it generated earlier. The 3MF attached was created using PrusaSlicer 2.5.0 by loading different STLs and preparing them for printing (e.p. added different box modifiers for infill and layers on the different models.
After saving the project and loading it up again the next day, I'm getting: "Loading of a model file failed."
[image]

I've read through similar existing issues: 
1. https://github.com/prusa3d/PrusaSlicer/issues/8292
2. https://github.com/prusa3d/PrusaSlicer/issues/7987

As far as I can tell this is a different issue: the file exists locally (1.) and contains what looks like valid data (although way too small given the imported 5+ STL were in the 2-5 MBs range and the 3MF project is < 1MB, might be truncated?)
Beginning of the file attached:
```
x00000000 50 4B 03 04 14 00 08 08 08 00 74 B1 BC 56 00 00 00 00 00 00 00 00 00 00 00 00 x`xPK........t..V............x
x0000001A 13 00 00 00 5B 43 6F 6E 74 65 6E 74 5F 54 79 70 65 73 5D 2E 78 6D 6C 85 90 DB xax....[Content_Types].xml...x
x00000034 4A C4 30 10 86 EF F7 29 42 6E A5 4D 55 10 91 B6 7B E1 E1 09 D6 07 18 D2 69 1B xaxJ.0....)Bn.MU...{.......i.x
x0000004E 4C 26 21 99 2E EB DB 3B D9 15 84 05 F5 72 E6 3F 7C F0 F7 FB 53 F0 EA 88 B9 B8 xaxL&!....;.....r.?|...S.....x
x00000068 48 83 BE 6D 3B AD 90 6C 9C 1C 2D 83 7E 3F BC 35 8F 7A 3F EE FA C3 67 C2 A2 C4 xaxH..m;..l..-.~?.5.z?...g...x
```

The project file can't be opened in Prusaslicer 2.6.0-beta3 either. It's likely that the file itself is corrupted, in which case the issue is that PrusaSlicer 2.5.0 produced the corrupted file. I am able to open all other 3MF files I tried.

### Project file & How to reproduce

Openthe following file with PrusaSlicer 2.5.0 or 2.6.0-beta3 on macOS:
[Dominion Card Holders.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11590045/Dominion.Card.Holders.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-macos

### Operating system

macOS Catalina 10.15.7

### Printer model

Prusa Mini+

johanLsp, 2023-05-29:
I confirmed the 3MF file is corrupted, it's not a valid zip file:
```
MacBook-Pro:Dominion$ unzip Dominion\ Card\ Holders.3mf
Archive:  Dominion Card Holders.3mf
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
```

Other 3MF files do unzip correctly. Still, PrusaSlicer 2.5.0 produced a corrupted 3MF file.

kocikdav, 2023-05-29:
Hello, thank you for reporting. You are right about the error displaying after unsuccessful export to 3mf - it does not propagate to UI. We will improve it. It seems to be logging to console with "error" level at least. But that's not how it should be.

FidelCapo, 2023-06-09:
Displaying error message when there is a problem with 3mf file export has been improved and the change will be part of the next release.
Closing.

closed 2023-06-09 by FidelCapo

uhlhosting, 2024-01-17:
Hi, I got the very same issue, on latest release of PrusaSlicer, and on MacOS Sonoma.

dkt0591, 2024-02-27:
I got same error, my working on it, how can i recovery ?

lukasmatena, 2024-02-28:
@uhlhosting @dkt0591 This can be completely different issue. Can you please create a new report (possibly referencing this one) and enclose the failing 3MFs? Thanks.