prusa3d/PrusaSlicer #10690: Ooze prevention logic should always print ooze shield immediatly after tool change

LightningShark25, 2023-05-29:
### Description of the bug

Printing multimaterial setup - 2 into 1, but hardware is irrelevant.

I'm using the Ooze prevention setting where I want the ooze shield to be printed after tool change, essentially using this shield to prime/clean the nozzle.

In some layers it seems the ooze shield will print prior to the toolchange, for that layer, the 'cleaning' part of the ooze shield isn't in effect.

Using a wipe tower has failed for me in the past, as with PVA it leaves strings of ooze across the print surface, causing adhesion issues.  The ooze shield printed, closer to the object has far less stringing issues

### Project file & How to reproduce

Print multimaterial
Set interface layer as second extruder only - soluble support
Set Oooze Shield 

[image]
[Body1.zip](https://github.com/prusa3d/PrusaSlicer/files/11591674/Body1.zip)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.2

### Operating system

Windows 10

### Printer model

Ender 3 with 2 into 1 hot end

kubispe1, 2023-05-30:
I hope I understand your hack, but it doesn't work properly, as I can confirm. However, I'm not sure if this hack is supported in general. What if there is missing material due to purging, and the single wall draft shield collapses for certain geometry combinations, either locally or globally? I'm not sure if it could bridge over. I think more testing needs to be done.

LightningShark25, 2023-05-30:
Thanks for the comment

I did not mention but the filament change is being done by purge bucket routine separately.   With wipe etc. The filament change happens off the bed 

The reason for still using ooze shield, it helps with extra cleaning action as I'm finding PVA just loves to ooze. Even during travel.

The integrity of the ooze shield doesn't really matter.. Its essentially just a wipe closer to the print.

LightningShark25, 2023-07-31:
Would really love if this could be considered as a feature or just an enhancement to the existing 'Draft Shield' function.

I have been testing this functionality where a single material per layer is used, and even with PVA, the integrity of the Draft Shield remains good.

Where it falls down is where there are multiple materials per layer (as per my example above). 

This could remove the need entirely for a purge tower, and give less stringing overall.
Please consider.

Thanks

LightningShark25, 2023-07-31:
[image]
[image]

Uploaded is an example of alternating layers of PVA and PLA..  Notice the 'Draft Shield' around the print has taken all the stringing and mess.. Once broken away the part inside is pretty clean for PVA

Lukrative525, 2023-12-22:
Has this been fixed? Can the issue be closed?

LightningShark25, 2023-12-29:
@Lukrative525 I'm not aware of any progress or anyone taking this up.
I never really got a decent approach to preventing stringing from PVA from messing up the integrity of prints and have abandoned it as support material for now...  I might revisit it one day

Lukrative525, 2023-12-29:
@LightningShark25 That makes sense. I only asked because I have basically your exact same setup, and I was messing around with this (haven't yet printed with draft shield though), and at least with my current settings, the draft shield _is_ printed first after a tool change (in PrusaSlicer 2.7.1).

I might take a look at your project file and see if there's some difference between our settings that makes this work for me but not for you. If I get brave enough, maybe I'll poke around in the source code and see if I there exists any logic for forcing this order.

Lukrative525, 2023-12-29:
@LightningShark25 Okay, well I can get your test file to slice as desired _if_ I enable the draft shield with 2 skirt loops.

closed 2026-09-29 by prusaslicer-bot[bot]