prusa3d/PrusaSlicer #10691: Outside perimeter 1/2 or smaller layer height

locki-cz, 2023-05-29:
Is your feature request related to a problem? Please describe.
Nicer outsite of prints, but still fast inside /what isnt vidible, easier print overhangs.

Describe the solution you'd like
Outside perimeter 1/2 of layer heigt or manual setup. Infill and inside perimeters stays normal layer height.

Describe how it would work
Print outside perimeters in 1/2 layer heigt, when prints 0,2 then it make two outside perimeters in 0,1 layer height. Probably not need set to print outside perimeters first... it must be tested.

neophyl, 2023-05-29:
Please check for duplicate issues both open and closed before creating new ones.  

For example https://github.com/prusa3d/PrusaSlicer/issues/8386  (which is a closed one as it was a duplicate of at least 4 other issues).  I think the earliest one for this particular request is https://github.com/prusa3d/PrusaSlicer/issues/402 first request in 2017.

sarusani, 2023-05-29:
https://help.prusa3d.com/article/infill_42#combine-infill-every-x-layers

Jan-Soustruznik, 2025-07-23:
Closed as duplicate of #8386