prusa3d/PrusaSlicer #10695: MMU painting workflow back to obj or parts

hoak, 2023-05-29:
**Is your feature request related to a problem? Please describe.**
I tried multicolor color painting and was surprised it worked as well as it did. When you paint a model with the tool in Prusaslicer the real magic happens when you slice the model after you finish painting.  Painting a model well can take a long time and is an art in itself. Seen in the slicing result, the slicing process creates actual objects assigned to different color/extruders; however these new part objects seem to only exist within the slicer memory and as far as I can tell are not exposed as real parts in the traditional sense such as when you add a multipart model and assign colors to objects.  Although the workflow now as designed is bug free, several enhancements could be added to make it even more capable and solve these problems when one tries to use this workflow for unintended results.  This causes several problems, which are all related to this being in memory and not on disk:
1. You cant use the color assignment parts outside of prusaslicer, such as in meshmixer or fusion360, since it the only workflow step possible is to then export to gcode.
2. You cant develop a set of objects independent of their use with Prusaslicers MMU painting workflow. To develop subsequent related models, remixes, etc that build upon the painting work.
3. A multipart model rather than a painted 3mf is much better to share on printables I would think and more flexible and easier for people to print and assign colors, as well as to change what parts are what color, and how many colors are used.
4. The paint session is fragile. Any change to the config like choosing another printer nozzle width wipes out the current painting.

**Describe the solution you'd like**
After painting a model, allow the in memory parts to be exported to obj parts on disk, so the model in slicer can then be reimported as real parts and subparts. Futhermore, rethink how we can have just one workflow where painting a single object, and assigning colors to multiple objects, share the same user experience.

**Describe how it would work**
Have an option in Prusaslicer to export a painted model as parts objects.

**Describe alternatives you've considered**
I have considered trying mesh mixer to create object sections of a mesh, but I like the paint tool in prusaslicer more, and am not sure how to get meshmixer to guarentee a certain depth of painted color into the interior of a model, like prusaslicer does well.

**Additional context**
These parts currenlty cannot be individually exported for instance as obj files. I wish they could because I could see using this tool as a real time saver to break up models into sections by color and then use these to make other combinations of the parts, and to preserve the painting color assignments. I prefer the color assignment workflow when you have actual parts. Even in Prusaslicer you lose all the painting work as soon as you make any major config changes.

ex-nerd, 2024-03-17:
This would be so useful. I had some filament break on a 5-color print and fail to print one color (would be nice if the XL could detect "no filament being extruded"). The "parts" for that color would actually print pretty well separately and I could have saved some time and plastic doing this and gluing them in.

mmcnair91, 2025-03-26:
this is a great idea!

ex-nerd, 2026-09-02:
Please keep this open (and add the feature request label). You're welcome to close it as "won't implement" (especially with the upcoming plugin system in PrusaSlicer 3) but I'd like that to come from an actual human not just some bot marking this as stale.