prusa3d/PrusaSlicer #10701: 3d connexion Space Mouse not working properly in 2.6.0

MikaelX, 2023-05-29:
### Description of the bug

In the beta I can only zoom in and out and pan side to side, the tilt function has gone missing

### Project file & How to reproduce

Enable SpaceMouse and try using it :) 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha6

### Operating system

MacOS 13.3.1 (a) (22E772610a)

### Printer model

Prusa MK3S+

foreachthing, 2023-05-30:
Works even on alpha3. You, probably, have to (re-)enable legacy support.
BUT, I'm on Windows, not Mac....
And I only have an **obsolete** SpacePilot - thanks, 3dconnexion!

[image]

MikaelX, 2023-05-31:
Tried both but no luck unfortunately :( Yeah 3dconnexion support we can discuss at a later stage :) I can try an earlier version to see if that works

MikaelX, 2023-05-31:
It seems to not work in 2.5.1 either, I will have to troubleshoot some. Works just fine in fusion 360 and other software...

MikaelX, 2023-05-31:
https://github.com/prusa3d/PrusaSlicer/issues/7277#issuecomment-1006359238 Found this from before, will try it

closed 2023-05-31 by MikaelX

MikaelX, 2023-05-31:
So.. I managed to solve it.

1. Downloaded & installed 3D connexion latest drivers from (march 2023)
2. Rebooted and nothing worked in prusa slicer
3. Disabled and re-enabled the support in prusa preference for 3d connexion, no luck
4. Rebooted again
5. Installed the plug-ins for 3d connexion (no idea if this was it or the next step)
6. Added PrusaSlicer as an application in 3d connexion software

Either step 5 or 6 fixed it, not sure but probably in combination with step 1.

[image]