prusa3d/PrusaSlicer #10708: Accessibility - many controls not labeled

SnowballsChance, 2023-05-31:
### Description of the bug

I support a blind user of purse slicer, he says that many of the controls are not labeled in such a way that his screen reader can identify the controls. He uses NVDA screen reader on Windows 10. 

### Project file & How to reproduce

Using a screen reader, attempt to find the save button when changing the filament settings

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.2

### Operating system

Windows 10

### Printer model

MK4

jfayre, 2023-07-04:
I am totally blind as well. Prusa Slicer has a bunch of accessibility bugs like this. For example, none of the checkboxes in the preferences window have proper accessible labels.
Additionally, tabbing through the UI will frequently get stuck. 
None of these should be particularly hard to fix. Someone just needs to take ownership of it.

m10d, 2024-08-21:
noticed another essentially identical serious accessibility bug here, just adding to raise visability https://github.com/prusa3d/PrusaSlicer/issues/12021

closed 2026-09-29 by prusaslicer-bot[bot]