prusa3d/PrusaSlicer #10713: 2.6 Beta 3 - Upload to Printer Host crashes PrusaSlicer

RAMilewski, 2023-05-31:
### Description of the bug

Upload to Printer Host crashes PrusaSlicer 2.6 Beta 3 when sending to Octoprint 1.9.0

From the MacOS Crash Report:

Process:               PrusaSlicer [68822]
Path:                  /Applications/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.6.0-beta3+MacOS-x64-202305261411 (PrusaSlicer-2.6.0-beta3+MacOS-x64-202305261411)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           PrusaSlicer [68822]
User ID:               501

Date/Time:             2023-05-31 14:01:00.820 -0700
OS Version:            Mac OS X 10.14.6 (18G103)
Report Version:        12
Anonymous UUID:        783EF29A-5F1D-0411-42C7-6A590B17F34F

Time Awake Since Boot: 150000 seconds

System Integrity Protection: enabled

Crashed Thread:        17

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x4] Symbol missing

Dyld Error Message:
  Symbol not found: _curl_url
  Referenced from: /Applications/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
  Expected in: /usr/lib/libcurl.4.dylib

### Project file & How to reproduce

[Bear Box Top 2_0.25mm_PLA_MINI_2h45m.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/11618932/Bear.Box.Top.2_0.25mm_PLA_MINI_2h45m.gcode.zip)

[image]

Clicking on either the UPLOAD or UPLOAD and PRINT button crashes Prusa Slicer.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 Beta 3  MacOS-x64

### Operating system

 Mac OS X 10.14.6 (18G103)

### Printer model

Prusa Mini

kocikdav, 2023-06-01:
We found the problem, no need for the log. Hopefully will be fixed in next beta.

RAMilewski, 2023-06-03:
Thanks!  Beta 4 fixes it for me!

closed 2023-06-03 by RAMilewski