prusa3d/PrusaSlicer #10715: Extrimely Long retractions every second layer

tonycstech, 2023-06-01:
### Description of the bug

I need old version of this program. Older then 2.5. Maybe 2.2 but i cannot find where to get one.
Every second layer (or so) retraction is 100mm or so for what ever reason.
Cannot print anything.
At 1st for what ever reason, wrong firmwre was selected, instead of marlin, it had reprap.
So i switched to marlin and it started to print as expected but on a 2nd layer, it retracted so much, filament came out from extruder all the way.

### Project file & How to reproduce

[Project.zip](https://github.com/prusa3d/PrusaSlicer/files/11621401/Project.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.2

### Operating system

W10x64

### Printer model

ROBO3D Original

lukasmatena, 2023-06-01:
@tonycstech I don't see any issue in the G-code, at least not when quickly looking over it. Could you please be more specific and point exactly to the line in the G-code that you think is incorrect? Thanks.

rtyr, 2023-06-01:
It is likely just misconfigured printer profile.

Based on your description, it seems to be related to absolute/relative extruder positioning. Maybe your FW expects relative positioning (it can be switched in PS).

Anyway, the g-code generated from your project file is OK for standard Marlin FW.

tonycstech, 2023-06-01:
I tried to switch absolute/relative but it gave me error so i was unable to even try it.

I looked at the gcode and i found this
1st layer

```
;LAYER_CHANGE
;Z:0.33
;HEIGHT:0.33
M117 STERRAD Tape Guide Tray
G1 Z.33 F9000
**G1 E-2 F2400
G92 E0**
```
2nd layer
```
;LAYER_CHANGE
;Z:0.66
;HEIGHT:0.33
M117 STERRAD Tape Guide Tray
G1 Z.66 F9000
**G1 E13.35186 F2400
G92 E0**
```

So every time there is a layer change, E is reset to 0 yet the following lines containing extrusion value do not start from 0 but rather continue to increase from last layer.
So after E0, next extrusion is G1 E13.35186 F2400

I never had this issue before. All i did was updated the program and all went south,
That is why i am asking for older version, 2.4-2.5
I am currently using 2.5.2 and there are conflicting settings.
Printer settings/Extruder 1/Retract on each layer change
Printer settings/Extruder 1/wipe while retracting
are the same as
Filament Settings/Filament Overrides
Two options are the same but do not seem to effect eachother.

rtyr, 2023-06-01:
Could you please share your g-code? The code you shared is different than output generated from your project file with your PS version.

This is g-code generated from your project file.
```
;LAYER_CHANGE
;Z:0.66
;HEIGHT:0.33
M117 STERRAD Tape Guide Tray
;WIPE_START
G1 F7200
G1 X91.436 Y8.185 E199.26217
G1 X90.738 Y8.185 E199.0413
G1 X91.99 Y9.437 E198.48058
G1 X91.99 Y10.135 E198.25972
G1 X90.762 Y8.906 E197.70962
;WIPE_END
G1 E197.60962 F2400
G92 E0
G1 Z.63 F9000
G1 Z.66
G1 X92.236 Y7.939
G1 E2 F2400
;TYPE:Perimeter
;WIDTH:0.5625
G1 F3600
G1 X92.236 Y68.938 E6.1148
G1 X6.765 Y68.938 E11.88045
```

> I tried to switch absolute/relative but it gave me error so i was unable to even try it.

If you mean error telling you to put G92 E0 to your layer-change g-code, than you can solve it by putting G92 E0 to layer-change g-code in printer profile.