prusa3d/PrusaSlicer #10716: Saving a modified preset with another name that matches an existing preset does not save the changes

awabom, 2023-06-01:
### Description of the bug

Say print setting presets 'Preset A' and 'Preset B' exists.
Modify 'Preset A', save with name 'Preset B'.

Expect modified values to be saved anyway, with the name 'Preset B', overwriting the existing 'Preset B'.

Instead, all changes are lost.

### Project file & How to reproduce

This requires two presets, so I don't think a project file helps?

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-beta3+win64

### Operating system

Windows 11

### Printer model

Ender 3 V2

awabom, 2023-06-01:
Also: If you have a modified print settings preset and select another preset - then press the 'Cancel' button in the dialog, the preset name still changes in the drop down list.

YuSanka, 2024-03-06:
Thanks for your report.
SPE-2188

FidelCapo, 2024-04-19:
This issue should be fixed and the fix will be part of the next release.
Closing

closed 2024-04-19 by FidelCapo