prusa3d/PrusaSlicer #10718: PrusaSlicer can not open 3mf files saved with Bambu Studio

markdesouza, 2023-06-01:
### Description of the bug

PrusaSlicer can not open 3mf files saved with Bambu Studio

### Project file & How to reproduce

See example file

[test.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11625214/test.3mf.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.2

### Operating system

MacOS 13.3.1

### Printer model

Not applicable

enricoturri1966, 2023-06-01:
This may help:
https://github.com/bambulab/BambuStudio/issues/169
?

garnos, 2023-08-29:
This still seems to be a problem. The programs listed on https://github.com/bambulab/BambuStudio/issues/169 still have issues opening a 3MF exported with Bambu Studio. Is there a known workaround for this?

cmguo, 2023-08-29:
I have a patch, https://github.com/prusa3d/PrusaSlicer/pull/10808

MightyPork, 2023-10-04:
It also can't open 3mf saved with FreeCAD anymore. I have to use STL instead

asiastuart, 2024-02-10:
Is there any chance this will be done in the near future? Half the multi color models on printables are now bambu originated and unusable.

lukasmatena, 2024-03-21:
Fixed in [PrusaSlicer 2.7.3-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.3-alpha1), thanks to @cmguo for the PR. Closing.

closed 2024-03-21 by lukasmatena

AlmostFeral, 2024-03-22:
While it's true that these files can now be opened in PrusaSlicer, I don't think this should have been closed. The intent is to be able to open colored .3mf files, and these are still coming through without the painting. The multi color models on printables that are bambu originated are still unusable.

engren, 2024-10-13:
Does not appear fully fixed. I downloaded the 3MF files from this project and they are able to open in Bambu but not with PS.

https://makerworld.com/en/models/684638

marcel151, 2025-05-04:
Any news? This is not fully fixed! Can't load 3mf from the following design:
https://makerworld.com/en/models/974431-rugged-darts-case-6-darts-target-swisspoint

raistlinJ, 2025-08-09:
I created a converter here: https://github.com/raistlinJ/3mf_bambu2prusa
Tested with several models and it seems to work; even transfers painted objects.
Hope it helps.