prusa3d/PrusaSlicer #10723: Filament presets disappear when Single Extruder multi material is enabled.

valkires1, 2023-06-02:
### Description of the bug

When you enable "Single Extruder multi material" even when a single extruder is selected the filament presets disappear. you can force some to reappear if you "Add/remove firmament" and add one, but once you close slicer they will disappear again till you uncheck that option. Additionally, if you do save a preset while the rest are hidden it will hide that preset as well.
I have experienced this on the public 2.5.2 and 2.6.0beta3

[image]
normal

[image]
enableing the setting

[image]
presets gone only one option

[image]
[image]
Closed and opened slicer now no options are available

[image]
[image]
selecting abs forced it to add abs but will disappear once I close and reopen program.

[image]
[image]
for some reason when restarted extruder 1 will show nothing but extruder 2 will show abs@mini

[image]
[image]
[image]
And once that is deselected issue instantly disappears

### Project file & How to reproduce

To recreate bug go to Printer settings -> general -> Single Extruder multi material
Note you will need to select more than one extruder to un-gray out box however once checked bug will occur with regardless if 1 or multiple extruders are selected.
Then go to Plater or filament setting to see problem, to worsen problem restart slicer. 
To disable bug, deselect "Single Extruder multi material" presets are restored instantly.
[Bug file example Test_Cube.zip](https://github.com/prusa3d/PrusaSlicer/files/11631954/Bug.file.example.Test_Cube.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-beta3+win64

### Operating system

Windows 10

### Printer model

i3mk3s+

rtyr, 2023-06-02:
> When you enable "Single Extruder multi material" even when a single extruder is selected the filament presets disappear.

rtyr, 2023-06-02:
> for some reason when restarted extruder 1 will show nothing but extruder 2 will show abs@mini

Maybe @YuSanka will want to check this, so I will leave this open at the moment.

valkires1, 2023-06-02:
Thank you, please do look into it.
However, if this is an expected result, this "future" is beyond frustrating, please consider removing so it behaves normally. It is very possible that I want to print at different filament setting, material types or different temperature at locations that could be used in SEMM. Perhaps I'm an idiot but I cannot see how this is not a net negative, what value dose that interaction add? it's very possible I'm an idiot and I don't understand the system.
Also forgive me for inputting my opinion if it's not appropriate in this setting.

rtyr, 2023-06-02:
We are talking about bundled system profiles meant for our standard printer profiles. You are using non-standard printer profile with "single extruder multi material" parameter enabled for some reason, even though this parameter is meant for MMU upgrade. We are hiding standard profiles on purpose for MMU and replacing them by MMU specific variants. You can see/change the condition on the "dependencies" tab.

rtyr, 2023-06-02:
> Maybe I'm also not getting it. But all my Filament profiles have "... and single_extruder_multi_material" in the "dependencies" field. But none of them are shown when I select "Single Extruder multi material" in the printer profile.

there is `!` in front of it

sarusani, 2023-06-02:
You're right, I'm an idiot 😄 
It's "and ! (... and single_extruder_multi_material)"

pigskins65, 2023-11-06:
@rtyr are you saying this issue is due to the MK3S+ system profile having that entry in the dependencies? I built a "single extruder MMU" type profile for the MK4 and it works great. Trying to create something similar for MK3S+ and am running into this same issue (all of my filament options disappear as soon as I check Single Extruder Multi Material). Can it simply be removed?

rtyr, 2023-11-06:
>  Can it simply be removed?

domme81, 2023-12-03:
Same issue here, very annoying! Not only for me, but also for a friend, who's still on the MK3S+.
With my new MK4 it works like intended, but as soon as I switch to my old MK3S, all filaments disappear from the dropdown lists, as soon as SEMM is switched on.

When a MK3 is selected, old profiles that were made for the MK3/S/+ shouldn't be ignored and disappear, but shown in the drop down lists. With the MK4 + SEMM option, it all works like intended.

I've been trying to report that issue a couple times since the last Slicer versions to the Prusa customer support, cuz I spent lots of nights to try to make it work! With my MK4 I'm now finally able to use that function, but on my MK3 no chance at all!! - PLEASE make a fix for that... To either somehow integrade the old profiles, that they don't disappear in SEMM mode, when a MK3/S/+ is selected, or make Slicer not ignore them, by checking which printer is selected and therefore, which filament profiles are displayed...

Can't be that hard to make a workaround, in my opinion.

Till now, all my requests for a fix over the last couple month got ignored. But there are quite a few people out there who also need a fix!

Thanks!

domme81, 2023-12-03:
I'm currently on PrusaSlicer 2.7.0, by the way. Waiting for a fix since 2.5.2, ...

domme81, 2023-12-03:
And they are compatible, because when I take out the SEMM function I can print with them like a charm. In normal mode the profiles work fine. So they must also work with SEMM switched on.

rtyr, 2023-12-03:
People tend to misunderstand SEMM parameter and probably using it for different purpose (it is also possible you don't have to enable it at all in your case). This parameter basically says you have our MMU unit or similar hardware. Those hidden profiles are not compatible with MMU, therefore are hidden. Anyway as I already mentioned here, bundled profiles are meant for standard hw and standard usage.

domme81, 2023-12-03:
Support: yeah, we fix that.... next Slicer version: still "broken" from the user's perspective.... Next Slicer version, same thing... Support: yeah, we forward that to the developer team.... Next time, same thing...

Please-pleasse think about how you can help the users getting that function to work again, with one of the next Slicer Versions... For those that are still in the MK3s, and just see the profiles disappear and don't know why and how to get them back and the function working again, like they were used to... before it suddenly stopped working for some unknown reaons....

domme81, 2023-12-03:
As soon, as S.E.M.M. is unchecked, the printer doesn't perform a filament change anymore, like with the S.E.M.M. hack, and keeps on printing with the same color, not stopping, not asking... Although different filaments are selected for the multiple tool heads.

Solution to the problem would be, if Slicer would add a M600 automatically in that mode (multiple toolheads selected, but no SEMM enabled) or an option to check "[x ] Multi materials / colors with a single extruder". Something like that...

If you have any other idea how to make it work without the hack that isn't originally meant for that, and stopped working for that other purpose since 2.5.2, please let me know. I'm aware, that I could edit the g-code, as long as it is the uncompressed one, but a more user friendly solution would be very appreciated. By me and I'm sure lots of others... If I still missed out on something, please let me know! :)

domme81, 2023-12-03:
You were absolutely right, the tool change box was blank for my current selected printer profile. I tested it with my MK4 standard profile that was just open, and just quickly changed the count of the toolheads, forgetting to add the M600... Gosh! :)

Now it's all working correctly. I also already changed my MK3 profiles, so now everything is fine! Finally!

I tried for the last month with my MK3 to get it running again, after it stopped working with the change to Slicer 2.5.2, but no chance. With the MK4 it suddenly worked again, although (in retrospective) SEMM was falsely enabled for that. Sorry that I didn't know better. Support couldn't help me out on that specific one - although they never disppointed me in any other cases - but that one was really frustrating! No solution, and I really spent a lot of time trying to fix that and find informations until today for the MK3... :) Now, with your explanations, it suddenly all makes sense!! :)

So: Thanks alot for getting that straight - you really made my day!! :)
Thank you for helping me out!! :)

rtyr, 2024-02-18:
I am closing this issue. Original issue is expected behavior.

closed 2024-02-18 by rtyr